PlotJuggler/PlotJuggler #244: Using a msg field as time-base (ROS)?

gavanderhoorn, 2020-01-25:
> Thanks for contributing to PlotJuggler.

Thank you for creating PlotJuggler :+1: 

PlotJuggler version: `2.5.0` (released into Melodic)

---

This is more of a question, not really an issue: I was wondering whether there would be a way in which a field in a ROS message would be usable as time-base for a plot (ie: for the X-axis). Or more generally: whether an arbitrary data source could be used as time.

With a message such as this one (example):

```yaml
Header header
uint64_t external_timestamp
float32 some_value
...
```

it's possible to create an XY-plot, with `external_timestamp` as the X and `some_value` for Y. This comes close, but PlotJuggler doesn't consider `external_timestamp` as "time", and additionally XY-plots don't get to use the nice time-derivative functionality.

This may be a question of RTFM, which would be fine, but after spending some time trying to figure this out I figured posting an issue here could be more efficient. If there is a better place to get support for these kind of questions, please let me know.

---

Working with external systems that generate data that gets transformed into ROS messages which don't set `header.stamp` to `external_timestamp` for whatever reason (clock drift, different time-bases, something else) but use `ros::Time::now()` can result in jittery plots, which makes it difficult to correctly interpret them (is the jitter coming from the external system/signal, is it a plotting artefact or is the ROS node doing something strange).

If `external_timestamp` could be used instead of `header.stamp` it would eliminate this problem.

gavanderhoorn, 2020-01-25:
PlotJuggler also often seems to want to auto-scale the Y-axis to values of `external_timestamp` (if that field ends up higher in the list) when creating XY-plots. Trying to set the scale for Y manually doesn't always seem to work, which, after switching X and Y axis (in the XY-plot dialogue) leads to heavy compression in the Y direction.

Not sure whether this would be a separate issue.

Regardless, it would be great if arbitrary datastream could be used as time-base.

facontidavide, 2020-01-31:
Short answer: no, it can't be done.

Long answer: 

In the general case, this is much harder than you may think. A timeseries is a vector of time/value pairs.

If I use another list of times, how do I find the corresponding value in the original timeserie? Linear interpolation between values? Closed value? 

"generating" values might "look good but the purpose of Plotjuggler is to analyze data, not to arbitrarily create one. I don't want to create artifact that might be misleading.

In **your particular case**, the new time seems to be part of the original message, so using XY plots might work, but, as you said, PJ is not prepared to address this.

A potential solution would be to create a more powerful way to create custom timeseries, i.e. improving considerably the Custom Functions engine.

I am afraid that I have hardly time for that :(

gavanderhoorn, 2020-01-31:
> In the general case, this is much harder than you may think. A timeseries is a vector of time/value pairs.
> 
> If I use another list of times, how do I find the corresponding value in the original timeserie? Linear interpolation between values? Closed value?

Are you assuming here that I'd be interested in plotting *other* data using this alternative time base?

That was not what I was thinking of, but I can see how that would be difficult.

I guess my idea was closer to XY plots, but then having plotjuggler recognise the X part as actual time, so all built-in functionality that works with time keeps working.

It would be nice if other messages with stamps deriving from the same time-base could be plotted, but that perhaps indeed difficult.

Right now I'm preprocessing the `.bag` and use a starting `header.stamp` which I then add to based on the delta-custom-`time_stamp` field.

facontidavide, 2020-01-31:
as I said, this particular case does have a solution, but in terms of User Interface, I don't know how to do it "elegantly".

The Custom Timeseries is the only place where this make sense, but it requires a considerable refactoring. I need to think about it...

facontidavide, 2020-02-08:
Good news. I have found potentially a way to solve this issue. I will notify you once it is done

facontidavide, 2020-02-08:
You can now optionally return a time/value pair instead of a value.
In this way you basically create a new time that is not related to the initial one anymore.

[image]

I will merge it with master soon

gavanderhoorn, 2020-02-08:
Very nice. :100: 

thanks for looking at this.

I'll be eager to test it out.

closed 2020-02-08 by facontidavide

gavanderhoorn, 2020-02-09:
https://github.com/facontidavide/PlotJuggler/compare/extended_editor hasn't been merged yet, has it?

facontidavide, 2020-02-09:
just merged