PlotJuggler/PlotJuggler #249: The principle of time offset function

Allenhanbo, 2020-02-12:
Hi Davide, thanks for your visualization tool, it help me a lot in visualization of rosbag. 

I'm doing research about quadrotor autonomous flying. I design the trajectory for quadrotor and then using rosbag to record the real trajectory of drone. Then I would like to analyze the trajectory accuracy(the error between the desired trajectory and real trajectory) in x, y, z separately. But I can not add my designed trajectory(it is a 7th order polynomial plotting in python) in PlotJuggler. So I'd like to use python to plot both my designed trajectory and trajectory of rosbag together. 

I face a problem now and I would like to hear your professional advice. I want to achieve the time offset function in python when insert the timeseries of rosbag in it, but I failed. Could you tell me how you can achieve the time offset function in PlotJuggler? I'd like to hear your advice?

facontidavide, 2020-02-12:
If you want to simply apply a time offset, this is your lucky day, because I added this feature 4 days ago:

https://github.com/facontidavide/PlotJuggler/issues/244

But in your case I would rather use a XY plot, since this would make easier to compare and time independent.

In your case you have the problem that your polynomial is not timeseries, by definition, but I understood from what you mentioned that you had found a workarounf for that.

Allenhanbo, 2020-02-12:
Hi David, thanks for your reply. As I mentioned above, I would like to plot the trajectory of rosbag myself with no help of plotjuggler, but I failed. When I was trying to using python to plot the trajectory x with timeseries of rosbag, I failed. I do not know how to do the time offset for timeseries(time start from 0), like you developed in plotjuggler. 

I know this is not the issue of plotjuggler, but I still hope if you can offer some help for my problem.

facontidavide, 2020-02-19:
I am not sure I can help :( Sorry

closed 2020-02-19 by facontidavide