PlotJuggler/PlotJuggler #250: ROS Subscriber plugin exception, type was not registered

JimmyDaSilva, 2020-02-19:
Hi @facontidavide !

I am using the ros Melodic debian version.
I get an exception when trying to suscribe to ROS messages from the `iiwa_msgs` package.
```
The plugin thrown the following exception:
This type was not registered
```
Do you know how I can solve that one ?

Many thanks,
Jimmy

JimmyDaSilva, 2020-02-19:
Multiple types if not every message type in `iiwa_msgs`.

Example of rosbag with type `iiwa_msgs/JointPosition` producing the exception:
[iiwa_bag.zip](https://github.com/facontidavide/PlotJuggler/files/4225819/iiwa_bag.zip)

[image]

Thanks !

facontidavide, 2020-02-19:
Ok, I will have a look

facontidavide, 2020-02-19:
I investigated and the problem is quite alarming. 

For some reason, the definition of the message is incomplete and does not correspond to the one you would find in your file **/opt/ros/melodic/iiwa_msgs/JointPosition.h**.

1) Just to double check, can you send me a copy of that file I mentioned?
2) Who is publishing this message? An embedded controller or a regular desktop application? Is **rosserial** involved somehow?
3) Is this problem recent of you always had that? Are other types OK?

JimmyDaSilva, 2020-02-20:
Oh, I see. It is indeed an alarming problem that could be explaining some communication issues I have been having with the robot.
The problem is probably caused by serialization because I do not have a problem when I publish the topic my self on my computer, and then look at the data with plotjuggler
(i.e., `rostopic pub /topic iiwa_msgs/JointPosition "...."` )

1/ Here is my iiwa_msgs/JointPosition.h
It is not installed in /opt/ros/melodic/... has is it cloned from this repo:
https://github.com/IFL-CAMP/iiwa_stack/tree/master/iiwa_msgs

2/ The IIWA controller is publishing theses messages with ros_java on a Windows OS.
I guess the compiled java version of this file is in this .jar file:
https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_ros_java/ROSJavaLib/iiwa_msgs-2.3.1.jar

3/ I just started using the IIWA, so I can't tell if it's a recent issue.
Some types work (i.e, `iiwa_msgs/MoveAlongSplineActionFeeback`) , others don't (`iiwa_msgs/CartesianPose`)
But the issue appears as well with ros "classic" message types coming from the robot, for example `actionlib_msgs/GoalStatusArray`

Here is a bag file with more topics if you want to have a look:
[bigger_iiwa_bag.zip](https://github.com/facontidavide/PlotJuggler/files/4230054/bigger_iiwa_bag.zip)

Thanks Davide !

facontidavide, 2020-02-20:
I am 90% sure that this is a problem with rosjava.

The good news is that I might now how to solve the problem. Bad news is that I don't have time to implement this right now.

JimmyDaSilva, 2020-02-21:
I understand. I will keep this opened then.
Thanks !

ndepal, 2020-03-19:
I have a potentially related issue (something about introspection):
When trying to stream a custom ROS message, PJ crashes with the following message:
```
/tmp/.mount_PlotJuMhC2gb/AppRun.wrapped: symbol lookup error: /tmp/.mount_PlotJuMhC2gb/usr/bin/libDataStreamROS.so: undefined symbol: _ZN16RosIntrospection7ROSTypeC1EN5boost16basic_string_refIcSt11char_traitsIcEEE
```
It apparently doesn't matter which message I try to stream, it also crashes with native types.

I'm using ROS Kinetic and PJ 2.6.2.

facontidavide, 2020-03-19:
@ndepal this is not related. Please check this other answer: https://github.com/facontidavide/PlotJuggler/issues/227

JimmyDaSilva, 2020-03-19:
Hi @facontidavide, after reading some issues on the IIWA_stack's side, I realized that the messages on the robot side are built rosjava Hydro, and my laptop runs on Melodic.
Source wiki: https://github.com/IFL-CAMP/iiwa_stack/wiki/addmessage

Possible fix to my problem:
Source issue: https://github.com/IFL-CAMP/iiwa_stack/issues/136

Feel free to close this issue or not depending on if you'd like to tackle these compatibility issues or not.

Cheers,
Jimmy

facontidavide, 2020-03-19:
I believe the problem is much more "profound" and not necessarily related to what you mention.

The way ROSJava serialize its message miss some information that PlotJuggler need.

The solution is to have a more graceful fallback to run-time information that I can read from command line.... but I am not planning to do this for the time being.

closed 2020-03-19 by facontidavide