PlotJuggler/PlotJuggler #253: Publishers/Start Rosout visualization stays checked after viz window is closed

ftsell, 2020-02-21:
# Description
When opening the /rosout visualization via the menu `Publishers/Start: ROS /rosout Visualization` a new window opens. When the window itself is closed via its window decoration (the x button), the menu item stays enabled.

# Expected Behavior
When closing the visualization window the menu item should disable itself automatically

# Steps to reproduce

1. Click on `Publishers`
2. Click on `Start: ROS /rosout Visualization`
3. Close the new window via its x button
4. Click on `Publishers`

The `Start: ROS /rosout Visualization` menu item is still enabled.

facontidavide, 2020-02-21:
I think I said some months ago "I should fix this".

Thanks to you I finally did ;)

closed 2020-02-21 by facontidavide

ftsell, 2020-02-22:
Thanks for fixing these issues so quickly. I greatly appreciate your work on PlotJuggler :D