PlotJuggler/PlotJuggler #257: Feature request: step backward while keeping robot status valid

rpvs73, 2020-03-02:
We have a configuration with PlotJuggler and RViz displaying a robot. When clicking on the time slider and then pressing left and right keys on the keyboard, it is possible to move one step backward or forward. 

However, for stepping backwards this requires two steps backward and one step forward to have RViz pick up the new simulation time and render the new state. When only stepping backward from PlotJuggler the robot status in RViz changes to 'Error' (caused by a tf time error).

In the attached movie, you can see that the status of the RobotModel changes to 'Error' when doing a number of steps back in time. The status will change back to 'Ok' on the first step forward.

Do you have a solution for this?

[tf_time_error.zip](https://github.com/facontidavide/PlotJuggler/files/4273677/tf_time_error.zip)

facontidavide, 2020-03-02:
Just to confirm... Are you setting use_sim_time?

rpvs73, 2020-03-02:
Yes, 'use_sim_time' is set to 'true'

facontidavide, 2020-11-23:
close due to inactivity and lack of interest

closed 2020-11-23 by facontidavide