PlotJuggler/PlotJuggler #261: PlotJuggler command not working for ROS

veimox, 2020-03-26:
When running PlotJuggler from the terminal if the package has been installed with the ROS version (`apt install ros-<distro>-plotjuggler`) it does not load the plugins.

I think the issues comes from how the plugins are loaded. Specifically here:

https://github.com/facontidavide/PlotJuggler/blob/master/plotter_gui/mainwindow.cpp#L126-L128

It automatically initializes plugins that are given by the path `QCoreApplication::applicationDirPath()` which [from the documentation](https://doc.qt.io/qt-5/qcoreapplication.html#applicationDirPath) I can see that in linux is the folder where the command is run. When doing `rosrun plotjuggler PlotJuggler` the folder is `/opt/ros/<distro>/lib/plotjuggler/` where the plugins are installed too, thus given the desired result. However when running `PlotJuggler` it is making use of the binary placed in `/opt/ros/<distro>/bin`, where there are no plugins.

I can think of some solutions:

1. we adapt the application to inject plugins in another way. e.g. looking for them in the LIBRARY_PATH or something like that.
2. creating at install time a static link. Either from bin to lib or reverse. Not sure how to do this...
3. making that the binary in `bin` is a file that executes the other binary. This would be simple and effective.

facontidavide, 2020-03-26:
I can not reproduce the issue. Running with `rosrun plotjuggler PlotJuggler` works just fine.

veimox, 2020-03-26:
Just run `PlotJuggler`

veimox, 2020-03-26:
That feature comes from #228

facontidavide, 2020-03-26:
yes, solution 3 is the most sensible one

closed 2020-03-26 by facontidavide