PlotJuggler/PlotJuggler #267: Fatal error: bfd.h: No such file or directory

hivemall, 2020-03-28:
cmake ..:
-- Could NOT find Bfd library named bfd.

make:
```
plotjuggler/3rdparty/backward-cpp/backward.cpp:27:
/home/user/aa_repositories/github__plotjuggler/3rdparty/backward-cpp/backward.hpp:198:12: fatal error: bfd.h: No such file or directory
 #  include <bfd.h>
            ^~~~~~~
compilation terminated.
make[2]: *** [plotter_gui/CMakeFiles/PlotJuggler.dir/build.make:776: plotter_gui/CMakeFiles/PlotJuggler.dir/__/3rdparty/backward-cpp/backward.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:520: plotter_gui/CMakeFiles/PlotJuggler.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
unsure what is "bfd"

jakob-andre, 2020-04-06:
Fixed this error when I installed this:  

 $ sudo apt-get install binutils-dev

Successfull build with Ubuntu 18.04 LTS and Ros Melodic
Best regards

closed 2020-04-06 by facontidavide

Kin-Zhang, 2021-03-25:
[image]
But I still have this problem:
```bash
The following packages have unmet dependencies:
 binutils-dev : Depends: binutils (= 2.30-15ubuntu1) but 2.30-21ubuntu1~18.04.4 is to be installed
                Depends: libbinutils (= 2.30-15ubuntu1) but 2.30-21ubuntu1~18.04.4 is to be installed
E: Unable to correct problems, you have held broken packages.
```
Then I just git clone from source it's ok

Kin-Zhang, 2021-03-25:
> [image]
> But I still have this problem:
> 
> ```shell
> The following packages have unmet dependencies:
>  binutils-dev : Depends: binutils (= 2.30-15ubuntu1) but 2.30-21ubuntu1~18.04.4 is to be installed
>                 Depends: libbinutils (= 2.30-15ubuntu1) but 2.30-21ubuntu1~18.04.4 is to be installed
> E: Unable to correct problems, you have held broken packages.
> ```
> 
> Then I just git clone from source it's ok

And Here is some detial I find on ubuntu forum:
https://www.ubuntuupdates.org/package/core/bionic/universe/proposed/binutils
WARNING: the "binutils" package was deleted from this repository