PlotJuggler/PlotJuggler #268: Is it possible to plot poses that are separated?

kibolho, 2020-03-28:
I have my bag file from a recorded trajectory, i am using "hector_trajectory_server" to get the trajectory. I would like to compare the planning trajectory and the real trajectory.

Thanks in advance!
[image]

facontidavide, 2020-04-15:
lot, lot of people wants to do this.

Unfortunately the answer is "you can't". This would require a new find of Plugin that generate a timeseries out of the list of points. Or the posibility to load in a XY plot something that is NOT a timeseries :(

Either way it is a bit of a hack. But there are so many people that wants to do that that I guess I must find a way... eventually.

But this will not happen anytime soon.

kibolho, 2020-04-15:
Hello!
I did a script in Matlab to solve my problem:
https://abilioazevedo.wordpress.com/2020/04/15/plot-trajectory-from-bag-file-using-matlab/
Thanks for answering!

closed 2020-04-17 by facontidavide