PlotJuggler/PlotJuggler #269: "show points and/or lines" option not saved in the layout file

orsoromeo, 2020-04-01:
First of all, thanks again for creating plotjuggler!

I am trying to plot multiple curves in the same plotjuggler plots that have the same name and an overlapping time series. This results in plotjuggler showing these different curves as if they were the same curve as in this figure:
[image]

This is fine with me, as I can work around this problem by simply plotting the points rather than the lines: 
[image]
However, I wish I was able to save my specific "show points and/or lines" option in the layout file. which is not the case at the moment. This means that whenever I start plotjuggler from scratch and I load my layout file the plot will look like in the first figure. I will then have to right click on each individual subplot in order to only show the points. 
In conclusion, It would save me a lot of time if I could save in the layout file the fact that I only want to show the points and not the lines.

facontidavide, 2020-04-15:
I will try to find some time to implement this

facontidavide, 2020-04-24:
done

closed 2020-04-24 by facontidavide