PlotJuggler/PlotJuggler #270: Segmentation fault (core dumped)

liuyf5231, 2020-04-07:
Hi,
I download the PlotJuggler-2.6.2-x86_64.AppImage. there may be some compatibility issues.

log is below, ubuntu 18.04 , 

```
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadROS.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamROS.so" : is a DataStreamer plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libRosoutPublisher.so" : is a StatePublisher plugin
"libRosTopicPublisher.so" : is a StatePublisher plugin
Stack trace (most recent call last):
#31   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Core.so.5", at 0x7f6b522b26ae, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#30   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f6b4ffd26db, in g_main_context_iteration
#29   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f6b4ffd264f, in g_main_context_dispatch
#28   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f6b4ffd2416, in g_main_context_dispatch
#27   Object "/tmp/.mount_PlotJuttcjsF/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5", at 0x7f6b480281ff, in QXcbGlIntegrationPlugin::qt_metacall(QMetaObject::Call, int, void**)
#26   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Gui.so.5", at 0x7f6b525b7f07, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#25   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Gui.so.5", at 0x7f6b525d41a4, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#24   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Gui.so.5", at 0x7f6b525d24e0, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#23   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Core.so.5", at 0x7f6b5225c26a, in QCoreApplication::notifyInternal(QObject*, QEvent*)
#22   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52bd8515, in QApplication::notify(QObject*, QEvent*)
#21   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52bd305b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#20   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52c32b7a, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#19   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52c3091c, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#18   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52bd7b31, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool)
#17   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Core.so.5", at 0x7f6b5225c26a, in QCoreApplication::notifyInternal(QObject*, QEvent*)
#16   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52bd8c18, in QApplication::notify(QObject*, QEvent*)
#15   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52bd305b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#14   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52d58ab2, in QMenu::event(QEvent*)
#13   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52c15fc7, in QWidget::event(QEvent*)
#12   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52d5805f, in QMenu::mouseReleaseEvent(QMouseEvent*)
#11   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52d540eb, in QMenu::internalDelayedPopup()
#10   Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52d4de51, in QMenu::setIcon(QIcon const&)
#9    Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52bcb897, in QAction::activate(QAction::ActionEvent)
#8    Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Widgets.so.5", at 0x7f6b52bc9411, in QAction::triggered(bool)
#7    Object "/tmp/.mount_PlotJuttcjsF/usr/bin/../lib/libQt5Core.so.5", at 0x7f6b5228aa8e, in QMetaObject::activate(QObject*, int, int, void**)
#6    Object "/tmp/.mount_PlotJuttcjsF/AppRun.wrapped", at 0x4aa569, in operator()
    | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject_impl.h", line 192, in impl
    | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h", line 552, in call<QtPrivate::List<>, void>
    | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h", line 495, in call
      Source "/home/df/ws_plotjuggler/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 487, in operator() [0x4aa569]
#5    Object "/tmp/.mount_PlotJuttcjsF/AppRun.wrapped", at 0x4aa0b3, in MainWindow::on_actionStartStreaming(QString)
      Source "/home/df/ws_plotjuggler/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 1235, in MainWindow::on_actionStartStreaming(QString) [0x4aa0b3]
#4    Object "/tmp/.mount_PlotJuttcjsF/usr/bin/libDataStreamROS.so", at 0x7f6b2d072f39, in DataStreamROS::start(QStringList*)
      Source "/home/df/ws_plotjuggler/src/PlotJuggler/plugins/ROS/DataStreamROS/datastream_ROS.cpp", line 353, in DataStreamROS::start(QStringList*) [0x7f6b2d072f39]
#3    Object "/tmp/.mount_PlotJuttcjsF/usr/bin/libDataStreamROS.so", at 0x7f6b2d08c46d, in RuleEditing::getRenamingRules()
      Source "/home/df/ws_plotjuggler/src/PlotJuggler/plugins/ROS/rule_editing.cpp", line 298, in RuleEditing::getRenamingRules() [0x7f6b2d08c46d]
#2    Object "/opt/ros/melodic/lib/libros_type_introspection.so", at 0x7f6b364a1e06, in RosIntrospection::ROSType::ROSType(boost::basic_string_ref<char, std::char_traits<char> >)
#1    Object "/opt/ros/melodic/lib/libros_type_introspection.so", at 0x7f6b364a1976, in 
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6b51434c40, in 
      Source "/build/glibc-OTsEL5/glibc-2.27/string/../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S", line 371, in  [0x7f6b51434c40]
Segmentation fault (Address not mapped to object [0x7ffdf7377040])
Segmentation fault (core dumped)
```

facontidavide, 2020-04-07:
It is the usual error: update **ros_type_introspection** and tell me if the problem is still there

liuyf5231, 2020-04-07:
> It is the usual error: update **ros_type_introspection** and tell me if the problem is still there

My system is the newest. but it still exists!

facontidavide, 2020-04-07:
ok, I will investigate

liuyf5231, 2020-04-08:
Hi facontidavide,
I tried 2.6.3, it is ok now!
Thanks.

closed 2020-04-08 by liuyf5231