PlotJuggler/PlotJuggler #271: Can't get custom function to work

msadowski, 2020-04-07:
Thanks for contributing to PlotJuggler. You are great!!!

# If you ignore these instructions, I will ignore your issue

Please allow me to better understand your problem providing the following information:

- I used version 2.6.2-1 (installed through apt-get for ros-melodic)

- Checked with example.bag from datafiles directory

I was trying to create a very simple custom function as follows:

[image]

However every time I'm trying to apply the transform by right clicking on the plot and selecting Custom... and then selecting Invert from the dropdown:
[image]

I get the warning:
"Something went wront while creating the following curves. Please check that the transform equation is correct.
"

[image]

I bet I'm doing something wrong but no matter what I don I can't seem to be able to apply the custom transform. Am I missing something?

facontidavide, 2020-04-07:
That looks like a bug!!! I will have a look at it. Thanks a lot for reporting

facontidavide, 2020-04-07:
Fixed. Thanks again

closed 2020-04-07 by facontidavide

msadowski, 2020-04-07:
Many thanks!