PlotJuggler/PlotJuggler #273: Feature Request: PlotJuggler 2.6.2. Time series to be able to reference the /clock topic for time, if it exists, rather than the wall clock.

githubfrond, 2020-04-15:
Thanks for creating PlotJuggler, You Are Great!
Excellent tool, just loving it.  I have an enhancement/feature suggestion for PlotJuggler.

Currently, PlotJuggler 2.6.2 appears to reference the wall clock no matter what the parameter use_sim_time is set to.  I hope I have this understanding correct?

When using ROS, it is not uncommon to do simulations that run faster than real time.  The usual method to do this is to set the use_sim_time parameter to TRUE and launch the nodes so they now use the automatically generated /clock topic as their basis for time.  A separate system is then used to publish simulation times to that /clock topic, driving the simulation.

It would be a great feature for PlotJuggler when run with ROS to use the same basis for time as all the other nodes launched from a file. Fortunately, you have created a very flexible tool, so my work around at this time is to add the /clock topic data as the X axis to each plot that I produce.

I hope the image of my nodes shows below the /clock topic being referenced.  Sorry for any lack of clarity on my part, happy to provide more code and examples if this isn't immediately understood.
 
[image]

facontidavide, 2020-04-15:
This is definitively something I should do. Hope to include this in 2.6.3

facontidavide, 2020-04-30:
Actually this was implemented a looooong time ago:

https://github.com/facontidavide/PlotJuggler/blob/master/plugins/ROS/DataStreamROS/datastream_ROS.cpp#L73-L79

Just tested it and it works perfectly using  `rosbag play --clock` and `rosparam set use_sim_time true`.

closed 2020-04-30 by facontidavide