PlotJuggler/PlotJuggler #274: Feature request: copy plot to clipboard

msadowski, 2020-04-20:
Hi Davide!

I was thinking that it would be a cool feature to allow copying the plot straight to clipboard instead of saving it to file.

Not sure how useful would this feature be for you or other users but I constantly use a screenshot tool to get the plot to my clipboard, this would definitely smooth out the workflow (at least for me).

Many thanks for your continued work on this project!

facontidavide, 2020-04-22:
It is probably easy...

https://doc.qt.io/qt-5/qclipboard.html

I will see when I can have a look at this

facontidavide, 2020-04-24:
[image]

Enjoy

closed 2020-04-24 by facontidavide

msadowski, 2020-04-24:
That's amazing, Thanks Davide!

facontidavide, 2020-04-28:
Now I am the one using this feature super often.

I am proud of myself 😂.

Seriously though, thanks for the idea and thanks for supporting PlotJuggler ❤️

msadowski, 2020-04-29:
You're welcome! Always happy to provide feedback. If you ever run Patreon or something similar to support development of PlotJuggler and BehaviorTrees.CPP then I'll definitely look into some long term support!