PlotJuggler/PlotJuggler #275: yaw_from_quaternion (and Roll and Pitch also)

WhiteBearSpirit, 2020-04-20:
Hello!
I see there's yaw_from_quaternion in the list of default transform functions, but it's not displayed in dropdown menu for series becuse it's considered 'invalid'.
I have written 3 working functions in Lua that do quaternion to Euler fransform.
They are not universal and are meant for my kind of data (.ulg flight logs from PX4 system)

QuaternionToRoll:
```
-- value is not used but should be mentioned to trick PJ check
z = $$vehicle_attitude/q.00$$
y = $$vehicle_attitude/q.01$$
x = $$vehicle_attitude/q.02$$
w = $$vehicle_attitude/q.03$$
t1 = 2 * (w * x + y * z)
t2 = 1 - 2 * (x^2 + y^2)
return math.atan(t1, t2)
```

QuaternionToPitch:
```
-- value is not used but should be mentioned to trick PJ check
z = $$vehicle_attitude/q.00$$
y = $$vehicle_attitude/q.01$$
x = $$vehicle_attitude/q.02$$
w = $$vehicle_attitude/q.03$$
sinp = 2 * (w * y - z * x)
pitch = 0
if math.abs(sinp) >= 1 then
    if sinp >= 0 then
        pitch = math.pi / 2
    else
        pitch = math.pi / -2
    end
else
    pitch = math.asin(sinp);
end
return (-1) * pitch
```

QuaternionToYaw:
```
-- value is not used but should be mentioned to trick PJ check
z = $$vehicle_attitude/q.00$$
y = $$vehicle_attitude/q.01$$
x = $$vehicle_attitude/q.02$$
w = $$vehicle_attitude/q.03$$
t1 = 2 * (w * z + x * y)
t2 = 1 - 2 * (y * y + z * z)
return math.atan(t1, t2)
```

More info can be found here: https://discuss.px4.io/t/pitch-roll-yaw-angle-estimated-data-in-ulog-files/16254/5

As you can see, I address data series directly with $$.
Is there a way to use 4 data series as an input for 1 function (feed them all to **value**)?
And PlotJuggler has funny validity check for functions: function should have a word **value** in it to be considered 'valid', that's why my functions have this commented line.

That is not really an issue, I just wanted to share these functions and maybe get an advice on how to 'beautify' them and make more usable for everyone.
Thank you!

facontidavide, 2020-04-22:
The drop down you mentioned menu is ONLY for functions that can transform a 1 timeseries into another timeseries.

The quaternion example is 4 timeseries input and 1 output, and this IS NOT SUPPORTED.

The "validity check" is of course a very quick and dirty test.

facontidavide, 2020-04-22:
[image]

Summarizing, this is a know limitation of the Function Editor: it works well with single-input, single-output, but it is cumbersome for multiple-inputs.

I don't think I will make any major improvement to this system for the time being...

closed 2020-04-22 by facontidavide