PlotJuggler/PlotJuggler #279: Data loading time is long when opening large bag

dvorak0, 2020-04-24:
Thanks for the awesome tool.

I compiled the master branch. And I open a large bag (~40Gb) with a single topic (sensor_msgs::Image), and I found it takes 300s to open the bag. I check the code and found in https://github.com/facontidavide/PlotJuggler/blob/master/plugins/ROS/DataLoadROS/dataload_ros.cpp#L238, the whole Image is read from disk.

Is there any possibility to optimize the loading time? I expect the actual disk read happens only when I play the data using timeserials bar.

Thanks.

facontidavide, 2020-04-24:
your day my friend, is about to get better.

https://github.com/facontidavide/PlotJuggler/commit/a99f85b3400d21363c70d91f50b7d91124d329b9

please let me know the new loading time

facontidavide, 2020-04-24:
Just a silly question... are you compiling in Release mode?
Please keep me updated about this, maybe, we can improve even further.

dvorak0, 2020-04-26:
Thanks for you reply. @facontidavide 

I am sure I compiled with -DCMAKE_BUILD_TYPE=RelWithDebugInfo.
I check the master branch e8784fffadf9c6a087d46477b8e56a269de613d1, with ros_comm at kinetic-devel.

I am not sure whether a99f85b3400d21363c70d91f50b7d91124d329b9 helps, as I found https://github.com/facontidavide/PlotJuggler/blob/e8784fffadf9c6a087d46477b8e56a269de613d1/plugins/ROS/RosMsgParsers/ros_parser.cpp#L122 already set the flag.

PlotJuggler renders Image/width Image/height as curves, which requires reading and deserializing the whole image data from disk.

facontidavide, 2020-04-26:
ok, my past self outsmarted my current self apparently.

Let's see if we can find (or not) a solution to this **together** (I need you to do some experiments for me,  since sharing the file seems unpractical).

1) There is the physical limitation that the entire 40 Gb MUST be read from disk, even if they are not parsed by PlotJuggler. This is a limitation in how data is stored in rosbags.

2) Ass you mentioned, it makes no sense for PlotJuggler to try parsing Images, pointclouds, etc. **Please de-select them** from the list of topics** that you want to load and tell me if it makes any difference.

3) as a reference/baseline, can you please tell me the time that on your PC **rqt_bag** takes to load the same file?

facontidavide, 2020-04-26:
FYI: 

about the line you mentioned, this is the only way it can be done. It is a limitation in the ROS interface that doesn't allow me (or anybody) to directly access the memory wit a zero-copy interface.

To be honest, that "write" command foes much more than you may think, since it also decompress the chunk of memory if the bag is compressed:

https://github.com/ros/ros_comm/blob/noetic-devel/tools/rosbag_storage/include/rosbag/message_instance.h#L168-L171

https://github.com/ros/ros_comm/blob/noetic-devel/tools/rosbag_storage/include/rosbag/bag.h#L403-L428

dvorak0, 2020-04-26:
Yes, let's do it together.

1. After de-select Image topics(only one topic with a few Images is selected), the loading is very fast(~4s).
2. rqt_bag costs 20s to open it. (I think only bag.open() is executed).

I understand the difficulty due to rosbag format. I also believe there is a way to optimize specifically for uncompressed Image data.

facontidavide, 2020-04-26:
Surely there could be, **forking rosbag_storage.** anyway, can we assume we solved the issue?

dvorak0, 2020-04-26:
I use plotjuggler to open images because I love the feature of republishing topics a lot. It gives perfect synchronization between images and curves.

Maybe we can provide a feature where the de-selected image topics can still be republished.

facontidavide, 2020-04-26:
Good news then. You should not need to load images to republishe them. Give it a try, if I remember it correctly, that is implemented already.

dvorak0, 2020-04-27:
Yes! You are right. All topics are republished even though I didn't select them.
Thanks. I am sure the issue can be closed.

closed 2020-04-27 by facontidavide