PlotJuggler/PlotJuggler #281: buildRosFlatType: There was an error parsing the buffer

ace-at-git, 2020-04-30:
Dear PlotJuggler's,

I have PlotJuggler 2.6.3 in an Ubuntu 18.04/ROS Melodic running. Opening the attached zipped ROSbag (recorded with a small diff drive bot) with
```
rosrun plotjuggler PlotJuggler -d 2020-04-30-11-57-06.bag
```
and selecting these signals:
```
/ros_car/left_wheel_angle_vel: std_msgs/Float32                     
/ros_car/left_wheel_cmd: std_msgs/Float32                     
/ros_car/right_wheel_angle_vel: std_msgs/Float32                     
/ros_car/right_wheel_cmd : std_msgs/Float32                 
```
...brings the following error message:
```
...
terminate called recursively
terminate called after throwing an instance of 'std::runtime_error'
  what():  buildRosFlatType: There was an error parsing the buffer.
Size 4 != 5, while parsing [/ros_car/right_wheel_angle_vel]
Aborted
...
```
After "filtering" the file with:
```
rosbag filter 2020-04-30-11-57-06.bag 2020-04-30-11-57-06_filtered.bag "topic != '/non_existend'"
```
all messages are still available. Using the new file like:
```
rosrun plotjuggler PlotJuggler -d 2020-04-30-11-57-06_filtered.bag
```
I can select and plot the mentioned signals without any problem!

The messages `/ros_car/[left | right]_wheel_angle_vel` causing problems are coming from a NodeMCU. The signals are simple standard messages with an non complex type. Rosserial is involved, this might also have an influence...!?

If I had a clue what exactly in the data structure/stream makes PlotJuggler unhappy I may get an idea about a possible cause in the source on the NodeMCU...

In case my systems/setup produces somehow malformed ROSbags, it would be nice if PlotJuggler would still be able to handle it direct without "filtering" them first.

[2020-04-30-11-57-06.bag.zip](https://github.com/facontidavide/PlotJuggler/files/4558264/2020-04-30-11-57-06.bag.zip)

facontidavide, 2020-04-30:
This is a known problem I had in the past with **rosserial**.

I never realized why, but I think it might be a good moment to fix this ;)

facontidavide, 2020-04-30:
fixed!

Thanks for reporting, I also found another bug you did not mention.

The application was supposed to show a nice error dialog, not to crash! Fixed that too

closed 2020-04-30 by facontidavide

ace-at-git, 2020-04-30:
Whow, that was quick - thank you very much!

facontidavide, 2020-04-30:
You are welcome

> Wow, that was quick

[image]