PlotJuggler/PlotJuggler #283: Reading big rosbag to RAM causes memory collapse 2.7.0

Andrew-NN, 2020-05-07:
Thanks for your software PlotJuggler, Davide!
Your previous version of software [2.6.3] can read big rosbags (approx 53gb) without any problems, which installed from official repo Ubuntu 16.04 x86_64.

I tried to build your latest update [2.7.0 master]. It works perfect, except reading big files, now it overloading RAM and crashes.

Build system: 
ROS kinetic catkin_make,
catkin 0.7.18,
cmake 3.5.1, 
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609,
QT: QMake version 3.0
Using Qt version 5.5.1 in /usr/lib/x86_64-linux-gnu

You can reproduce this bug with any large rosbag (file size more than your's PC's RAM), if you'll try to load it completely.

[image]

P.S.: if it is not bug, please tell me where to look & fix this "RAM-loading" feature.

Gratefully and with best regards,
Andrew

facontidavide, 2020-05-07:
Weird... Thanks for reporting

facontidavide, 2020-05-07:
In my benchmark 2.6.3 and 2.7.0 use the same amount of RAM... too much.

But I just pushed a change that improves it. https://github.com/facontidavide/PlotJuggler/commit/8be0b0a373b961cf51a4fad56793fd94e6007642

Give it a try.

Andrew-NN, 2020-05-07:
Thanks for reply, Davide!

I rebuild it with last commits, but result the same.

Originally rosbag filesize 53 gb approximately. 

[image]

facontidavide, 2020-05-07:
This is extremely weird... I am quite positive that they are supposed to use the same amount of RAM.

PlotJuggler is supposed to load everything in RAM, and it uses 16 bytes per each data point (with a lot of refactoring, that can be reduce to almost halph, but this is another story).

With 53 Gb I actually expect **both** version to crash. I don't understand how 2.6.3 can possibly work.

Can you please split the rosbag with **rosbag_editor** (a software of mine, google for it) and send me a 1 gb sample? if you don't want to share it here, send me a WeTransfer link at my email davide.faconti@gmail.com

May I ask for which application are you using PlotJuggler? Just curious.

facontidavide, 2020-05-07:
One more test. Instead of using the AppImage, can you please switch to tag 2.6.3 in the git repository

> git checkout 2.6.3

And tell me if the problem actually goes away and how much RAM is used? 
You mentioned that you are compiling in Debug. What if you compile in Release? It should not be different, but... something is different, clearly.

Andrew-NN, 2020-05-07:
Shure, I'll try with tags. Thank you!

Debug/Release 2.7.0 does not matters, I tried it before.

Andrew-NN, 2020-05-07:
> This is extremely weird... I am quite positive that they are supposed to use the same amount of RAM.
> 
> PlotJuggler is supposed to load everything in RAM, and it uses 16 bytes per each data point (with a lot of refactoring, that can be reduced to almost **half**, but this is another story).
> 
> With 53 Gb I actually expect **both** version to crash. I don't understand how 2.6.3 can possibly work.
> 
> Can you please split the rosbag with **rosbag_editor** (a software of mine, google for it) and send me a 1 gb sample? if you don't want to share it here, send me a WeTransfer link at my email [davide.faconti@gmail.com](mailto:davide.faconti@gmail.com)
> 
> May I ask for which application are you using PlotJuggler? Just curious.

Yeah of course. I research self driving car monitoring cases in NNSTU n.a. R.E. Alekseev (University). Your application is very helpful for my research. 

I'll send to you 1gb in private.

Andrew-NN, 2020-05-07:
Unfortunately, it cause by new version 2.7.0.

I rebuilt with 2.6.3, and we can see the difference:
[image]

Andrew-NN, 2020-05-08:
Thank you again, Davide. 
Now, I guess that file no needed.
I'll try to get reason of this glitch, and report to you.

--
Best wishes,
Andrew

closed 2020-05-08 by Andrew-NN

facontidavide, 2020-05-08:
Thank you for helping me identifying this "problem" that I did not know to have.

My files are too small ;)