PlotJuggler/PlotJuggler #286: Feature Request: Make available string fields from ROS topics in function source data

alunawat, 2020-05-13:
If you can make available string fields in the function source data field, we will be able to manipulate string fields to numeric fields for plotting. 

I just want to acknowledge how great this tool is and I've been recommending this to everyone who it is relevant to. Thanks again for developing and maintaining this.

facontidavide, 2020-05-13:
I think that can be done, but the Function Editor needs some serious refactoring because I am not 100% happy with its usability for complex cases... like yours.

This change is harder than you may think because right now the string fields are discarded immediately.

Which message are you thinking about? Is it a custom one?

alunawat, 2020-05-14:
Yes it is a custom one. I use a string field to encode hex numbers as identifiers for a custom sensor message. It would be awesome if I could convert that using code in the function editor and plot it on PlotJuggler.

facontidavide, 2020-05-14:
this requires a lot of changes. I think it is a good feature that may help many users, but it will takes me time to do it

Ellon, 2021-07-15:
Hello @alunawat, @facontidavide 
+1 for this feature. :) 

I have a use case where I use a custom message to send some "details" about the system, and I would like to filter which data to plot based on a string field of this message.

maracuya-robotics, 2021-12-17:
Somehow related to this is my situation: I have a CSV file with fields consisting of strings. I'd like to manipulate these strings (shown under `value` in the data tree) with custom timeseries and return a number. It seems that strings are not accepted as input in custom timeseries, as even a constant return doesn't produce any output.

nicholas-p1, 2022-06-24:
+1 for this feature

I have a state machine that publishes the name of the current state. It would be useful to visualise the current state next to a sensor plot to know what the robot is doing at that moment and enable better/easier interpretation of the sensor values

PepMS, 2022-12-20:
+1 for this feature

same case as @nepfaff

facontidavide, 2026-10-01:
Closing this PJ3 report about using string fields as inputs to custom numeric functions during backlog cleanup: it is over three years old, and there is no current implementation/reproducer follow-up in this thread that warrants keeping it in the active queue. Development now focuses on PJ4 (`main-4.x`). If this remains needed with a current PJ4 build, please open a focused PJ4 issue describing the remaining behavior; this discussion is retained for reference.

closed 2026-10-01 by facontidavide