PlotJuggler/PlotJuggler #289: What is the status of the ROS2 branch?

mlautman, 2020-05-19:
Heyo, I wanted to ask about the state of the ROS2 branch. Is it intended for public consumption yet? 

It built locally but it doesn't source correctly which (`ros2 pkg list` doesn't pick it up). After looking through the CMakeLists.txt, it seems that you intend for this branch to eventually support both ROS and ROS 2. This makes me think that the branch might still be a work in progress and will eventually be merged into master. Is this the case?

Thanks for taking the time!

facontidavide, 2020-05-19:
just added a commit. see if that helps

mlautman, 2020-05-19:
Thanks for the fast reply :) 

I'll take a look and see if I can find the issue.

mlautman, 2020-05-19:
I'm not sure I see a new commit. Newest commit is 285f2a575c8357b13a109efbf701da0d04465a14 from 16 days ago

facontidavide, 2020-06-13:
ros2 support is ready for rock and roll baby (eloquent only)

closed 2020-06-13 by facontidavide