PlotJuggler/PlotJuggler #291: Segfault when working with ROS 2 eloquent generated messages

mlautman, 2020-05-20:
This issues builds off of #289 and #290. 

When running plotjuggler with ROS2 eloquent on a 18.04 machine, I get the following when trying to  use the ros2 stream:
```
#4  | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject_impl.h", line 168, in 
    |   166:                 break;
    |   167:             case Call:
    | > 168:                 FuncType::template call<Args, R>(static_cast<QFunctorSlotObject*>(this_)->function, r, a);
    |   169:                 break;
    |   170:             case Compare: // not implemented
    | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h", line 240, in 
    |   238:         template <typename SignalArgs, typename R>
    |   239:         static void call(Function &f, void *, void **arg) {
    | > 240:             FunctorCall<typename Indexes<N>::Value, SignalArgs, R, Function>::call(f, arg);
    |   241:         }
    |   242:     };
    | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h", line 130, in 
    |   128:     struct FunctorCall<IndexesList<II...>, List<SignalArgs...>, R, Function> {
    |   129:         static void call(Function &f, void **arg) {
    | > 130:             f((*reinterpret_cast<typename RemoveRef<SignalArgs>::Type *>(arg[II+1]))...), ApplyReturnValue<R>(arg[0]);
    |   131:         }
    |   132:     };
      Source "/home/mike/ws_ros2/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 487, in operator() [0x55d0f831299b]
        484:                     ui->menuStreaming->addSeparator();
        485: 
        486:                     connect(startStreamer, &QAction::triggered, this,
      > 487:                             [=](){  on_actionStartStreaming(plugin_name); });
        488: 
        489:                     connect(streamer, &DataStreamer::connectionClosed,
        490:                             ui->actionStopStreaming, &QAction::trigger );
#3    Source "/home/mike/ws_ros2/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 1235, in MainWindow::on_actionStartStreaming(QString) [0x55d0f8312521]
       1232:     bool started = false;
       1233:     try{
       1234:         // TODO data sources
      >1235:         started = _current_streamer && _current_streamer->start( nullptr );
       1236:     }
       1237:     catch(std::runtime_error& err)
       1238:     {
#2    Source "/home/mike/ws_ros2/src/PlotJuggler/plugins/ROS/DataStreamROS2/datastream_ros2.cpp", line 127, in DataStreamROS2::start(QStringList*) [0x7f277330d5a8]
        124:     for (const auto& topic : dialog_topics) {
        125:         if( _config.selected_topics.contains(topic.first) )
        126:         {
      > 127:             subscribeToTopic(topic.first.toStdString(),
        128:                              topic.second.toStdString());
        129:         }
        130:     }
#1  | Source "/home/mike/ws_ros2/src/PlotJuggler/plugins/ROS/DataStreamROS2/datastream_ros2.cpp", line 181, in DataStreamROS2::subscribeToTopic(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    |   179:     if( topic_info_it == _topic_info.end() )
    |   180:     {
    | > 181:         topic_info_it = _topic_info.insert( {topic_name, TopicInfo(topic_type)} ).first;
    |   182:     }
    |   183:     TopicInfo* topic_info = &(topic_info_it->second);
      Source "/home/mike/ws_ros2/src/PlotJuggler/plugins/ROS/ros2_parsers/ros2_parser.h", line 25, in DataStreamROS2::subscribeToTopic(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) [0x7f277330bafc]
         23:         introspection_support = rosbag2::get_typesupport(type, introspection_identifier);
         24:         type_support = rosbag2::get_typesupport(type, typesupport_identifier);
      >  25:         has_header_stamp = Ros2Introspection::TypeHasHeader( introspection_support );
         26:         buffer = rosbag2::allocate_introspection_message(introspection_support, &allocator);
         27:     }
#0    Source "/home/mike/ws_ros2/src/PlotJuggler/plugins/ROS/ros2_introspection/src/ros2_introspection.cpp", line 88, in Ros2Introspection::TypeHasHeader(rosidl_message_type_support_t const*) [0x7f2773318e95]
         85:     if( members->member_count_>=1 )
         86:     {
         87:         const MessageMember& first_field = members->members_[0];
      >  88:         const auto* header_members = static_cast<const MessageMembers*>(first_field.members_->data);
         89:         if( strcmp( header_members->message_name_, "Header") == 0 &&
         90:             strcmp( header_members->message_namespace_, "std_msgs::msg") == 0)
```
And when I try to load a ros2 bag I get the following segfault:
```
#6    Source "/home/mike/ws_ros2/build/plotjuggler/plotter_gui/PlotJuggler_autogen/EWIEGA46WW/moc_mainwindow.cpp", line 481, in MainWindow::qt_metacall(QMetaObject::Call, int, void**) [0x55cc51346c48]
        478:         return _id;
[... 38 lines cut ...]
    |   107:         if( _topic_info.count(topic_name) == 0 )
    |   108:         {
    | > 109:             _topic_info.insert( {topic_name, TopicInfo(topic_type)} );
    |   110:         }
    |   111:     }
      Source "/home/mike/ws_ros2/src/PlotJuggler/plugins/ROS/ros2_parsers/ros2_parser.h", line 25, in DataLoadROS2::readDataFromFile(FileLoadInfo*, PlotDataMapRef&) [0x7f290f30878f]
         23:         introspection_support = rosbag2::get_typesupport(type, introspection_identifier);
         24:         type_support = rosbag2::get_typesupport(type, typesupport_identifier);
      >  25:         has_header_stamp = Ros2Introspection::TypeHasHeader( introspection_support );
         26:         buffer = rosbag2::allocate_introspection_message(introspection_support, &allocator);
         27:     }
#0    Source "/home/mike/ws_ros2/src/PlotJuggler/plugins/ROS/ros2_introspection/src/ros2_introspection.cpp", line 88, in Ros2Introspection::TypeHasHeader(rosidl_message_type_support_t const*) [0x7f290f310be5]
         85:     if( members->member_count_>=1 )
         86:     {
         87:         const MessageMember& first_field = members->members_[0];
      >  88:         const auto* header_members = static_cast<const MessageMembers*>(first_field.members_->data);
         89:         if( strcmp( header_members->message_name_, "Header") == 0 &&
         90:             strcmp( header_members->message_namespace_, "std_msgs::msg") == 0)
         91:         {
Segmentation fault (Address not mapped to object [0x8])
```

An initial parsing of this seems to indicate that there is an issue parsing the message type info which makes sense because the IDL has changed quite a bit in recent releases of ROS.

My question is, 1. Which ROS2 distro does the ros2 branch target? and 2)

mlautman, 2020-05-20:
This can be reproduced by running: PlotJuggler in one terminal and the following command in another:
`ros2 topic pub -r 5 /my_topic std_msgs/msg/Int16 "data: 5"`

mlautman, 2020-05-20:
fixed with new commit in #290

facontidavide, 2020-05-20:
you are looking at an obsolete branch. On master that was already solved.

https://github.com/facontidavide/PlotJuggler/blob/master/plugins/ROS/ros2_introspection/src/ros2_introspection.cpp#L40-L43

closed 2020-05-20 by facontidavide