PlotJuggler/PlotJuggler #293: Plot data with websocket plugin

bgromov, 2020-05-22:
**PlotJuggler**: 2.6.4
**Platform**: macOS 10.15.4 (Catalina)

I am trying to visualize data using the websocket plugin. The server connects and apparently receives the data (I can see the time running and it corresponds to the time values I send), but I do not see any data streams listed on the left pane and therefore can't plot them. Do I need to configure something else or is it a bug?

I am streaming data using the format from the example client, i.e.: `<key>:<time>:<value>`.

(Interestingly, with the example client I am able to see only the "cos" data stream, while the client is actually also publishing the "sin" data.)

What could be the problem?

Thanks

bgromov, 2020-05-22:
Sorry for the false alarm. Clean build solved the issue.

bgromov, 2020-05-22:
No, there is a problem indeed. Rebuilding doesn't help anymore.

I went on a bit of debugging and can tell that the plugin receives the data. At least these lines are fine, the new numerics are added and the data is pushed:
https://github.com/facontidavide/PlotJuggler/blob/b8f7f62d1cd2274ec3793d42fef00584b811afe9/plugins/DataStreamWebSocket/datastreamserver.cpp#L99-L106

But for some reason PlotJuggler doesn't show the streams in the list... Well, It does show some of them, but not all. For example, these appear:

```
drone/x
drone/y
drone/z
drone/yaw
```

and these don't:

```
target/x
target/y
target/z
target/yaw
```

Any ideas?

bgromov, 2020-07-16:
I found a workaround for now: If I drag an empty list of streams to the plot and then do "Delete everything", the data streams appear.

This is also valid for the latest version 2.8.3

facontidavide, 2020-07-17:
1. I have low interest in this plugin for the time being. It was contributed by another person with a PR that I ccepted, but for the time being I am focusing on something else.

2. How can I repdroduce this???? Can you send the code of the message publisher to make my life easier, if I ever decide to address this?

3. Consider solving the issue yourself and submit a PR.

4. If option 3 is not available, consider become a Github sponsor to get this issue prioritized.

bgromov, 2020-07-18:
> 2. How can I repdroduce this???? Can you send the code of the message publisher to make my life easier, if I ever decide to address this?

Here is the minimally (non-)working example (requires Python 3):

```python
#!/usr/bin/env python3

import websockets
import asyncio
import math

async def send():
    async with websockets.connect('ws://127.0.0.1:6666') as ws:
        for i in range(0, 10000):
            await ws.send(f'cos:{i/10}:{math.cos(i/10)}')
            print(f'> cos:{i/10}:{math.cos(i/10)}')
            await asyncio.sleep(0.1)

asyncio.run(send())
```

To reproduce, start PlotJuggler and enable streaming with `Streaming -> Start: WebSocket Server`, then run the above script in the terminal.

In the PlotJuggler you will see that the time is running but the left pane is empty. Left-click on it (although it is empty) and drag it to the plot area, then delete everything (`Data -> Delete everything`) and the `cos` stream should appear in the list. It may require to do a couple of times.

It seems also to depend on the data rate, e.g. when I set the delay to 1 second, there is no way I can make the stream to appear on the list.

> 3. Consider solving the issue yourself and submit a PR.

Could you give any hints on where to start with this? Since the above manipulations help sometimes to get the streams showing, I would expect it is somehow related to the GUI...

Thanks!

facontidavide, 2020-07-18:
The problem is most definitely related to the plugin, not the GUI. I will have a look this weekend

facontidavide, 2020-09-29:
My plan is to soon-ish rewrite completely the plugin to support more formats (JSON, INI, TOML, YAML) and more transports (Websocket, MQTT, UDP, etc). Sorry.

Therefore, for the time being, I am not focusing on this issue.

facontidavide, 2020-11-23:
refactoring done

closed 2020-11-23 by facontidavide