PlotJuggler/PlotJuggler #294: Uncaught exception: X and Y axis don't share the same time axis

bgromov, 2020-05-22:
PlotJuggler crashes with the following error after some time:

    libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: X and Y axis don't share the same time axis

I am streaming data using the websocket plugin, so the time values arrive in text, and I am sure they are completely identical.

This happens if and only if the buffer length is set (tried only setting it to 60 sec though). Looks like some of the values that should have a pair for XY plot are getting dropped because the buffer is trimmed, so the pairs break.

Btw, is the buffer length infinite by default? If I don't change the value in the bottom right spinner the data seems to accumulate forever.

Thanks

facontidavide, 2020-06-13:
- you should use the latest PlotJuggler. 2.6.4 is very old
- GIVE me a WAY to REPRODUCE THE ERROR

facontidavide, 2020-06-13:
> Btw, is the buffer length infinite by default? 

I will check that, it should't happen

facontidavide, 2020-06-13:
can not reproduced in latest version

closed 2020-06-13 by facontidavide

facontidavide, 2020-07-07:
fixed in version 2.8.2