PlotJuggler/PlotJuggler #296: cannot creat folder at `home`

lianghongzhuo, 2020-06-12:
Hi,
In our system, ` /home` folder is blocked by the file system manager to mount remote users. 
So when we install plotjuggler by the command `sudo apt install ros-XX-plotjuggler`, we will get an error:
```
 error creating directory './home/buildfarm': Permission denied
```
Could you put the `buildfarm` somewhere else?
For example, our typical home folder is located at `/homeL`
So by detecting the `$HOME` value, one can know where to create the `buildfarm` folder.
Thank you.

v4hn, 2020-06-12:
No system binary is supposed to install any files in `/home` or any `$HOME` for that matter.
This is clearly a bug in the released packages due to short comings in handling the icon and desktop files [which I complained about before](https://github.com/facontidavide/PlotJuggler/commit/e8784fffadf9c6a087d46477b8e56a269de613d1#r38730664).

@facontidavide Please disable the logic for buildfarm builds. As this is still not standard-conform, I would propose to drop it again altogether, possibly providing the files for users to copy on their own if they want them around.

facontidavide, 2020-06-12:
Yes I will just drop this altogether

commit fe38f930 (2020-06-13): fix #296
closed 2020-06-13 by facontidavide