PlotJuggler/PlotJuggler #297: Zoom doesn't work for me

cathrinS, 2020-06-19:
Hi,

I really enjoy using PlotJuggler, regrettably zooming isn't working for me at all. I've read the instruction on the webpage:
```
Zoom
To zoom a specific area, click Left Mouse Button on a plot and move the mouse keeping the button pressed.
Alternatively, use the Mouse Wheel or the the buttons Plus(+) or Minus(-).
```
but none of them works. There isn't any hint related to this in the Terminal-Window. Any suggestion what might be the reason? Could it be that I run it in a VM?

I use version 2.8.1, OS: Ubuntu 18.04 (in VMWare)

cathrinS, 2020-06-19:
After starting once with `plotjuggler -t`, plotjuggler works afterwards as expected! No clue what changed

facontidavide, 2020-06-19:
no idea :(

cathrinS, 2020-06-19:
Thanks for your fast reply, just realized that this problem came up again, probably it has to do something with the ROS-Subscriber I use all the time. The small "legend" depicting the values of a time-series-point doesn't work, too. I'll reopen the issue as soon as I can reproduce the steps leading to the problem

cathrinS, 2020-06-22:
Ok, reproducing was easier than I thought: 
Zooming works if I load a bag into PlotJuggler. As soon as I start a ROS-Subscriber, zooming doesn't work anymore. It works again as soon as I stop the subscriber. I think this is not a bug but intended behavior, isn't it?
I often subscribe to 4 different topics out of >10 and I would prefer to be able to zoom into the plot without the need to stop the subscriber. Otherwise I would need to stop and start the subscriber for every zoom-operation.
"Zoom" refers to: zoom in, the vertical red line (otherwise just the points get a little bit larger by using Shift + Left-Mouse) and the "y-value-legend" next to the closest point.

aeudes, 2020-06-22:
You can also introspect the data (zoom) when streaming is off : their is a convenient shortcut for that ctrl+space.

facontidavide, 2020-06-22:
ok, now it is clear. THAT is intended behaviour.

Zoom will work only when streaming is off

closed 2020-06-22 by facontidavide