PlotJuggler/PlotJuggler #300: Segmentation fault when subscribing to gazebo topics

ljburtz, 2020-07-02:
Hello and thanks for this great program.
I have an issue plotting data from gazebo9.

### Issue A: Seg fault when subscribing to the live gazebo simulation
`Segmentation fault (Address not mapped to object [(nil)])`

Subscribing to gazebo topics leads very quickly (few seconds) to a segmentation fault / core dumped : attached log
[plot_juggler_fromSource.log](https://github.com/facontidavide/PlotJuggler/files/4862860/plot_juggler_fromSource.log)
[plot_juggler_debianPackage.log](https://github.com/facontidavide/PlotJuggler/files/4862884/plot_juggler_debianPackage.log)
(the logs are lauching Plotjuggler with a layout but Plot juggler also crashes if I just start it without any cli arguments and then manually choose the topics)

### Issue B: Seg fault when subscribing to a rosbag made from the gazebo simulation:

- I recorded the ROS messages : recording started before launching plotjuggler, recording stopped after plotjugler crashed. (attached as .bag).

[segfault_2020-07-02-12-08-07bag.zip](https://github.com/facontidavide/PlotJuggler/files/4862867/segfault_2020-07-02-12-08-07bag.zip)

- Then shutdown gazebo. 
- Then restart roscore. 
- Then `rosbag play -l segfault_2020-07-02-12-08-07.bag`. 
- Then launch plotjuggler: (for example with this layout: [plotjuggler_groundruth_xml.zip](https://github.com/facontidavide/PlotJuggler/files/4862903/plotjuggler_groundruth_xml.zip))  

There is no crash like in issue A. 
But after a while (several minutes) there is the same seg fault and error output as reported in #93 
(I am also using XY plot of messages in the same topic). 
log attached: 
[plot_juggler_XYsame_topic.log](https://github.com/facontidavide/PlotJuggler/files/4862874/plot_juggler_XYsame_topic.log)
` throw std::runtime_error("X and Y axis don't share the same time axis");`

I could not reproduce the issues with datasamples/example.bag
I wonder if there is a race condition caused by the messages being published at very high hz and bw by gazebo.

Environment: 
- Using v2.8.1 compiled from source / master branch (latest commit: 6364f73698f827b02af41414a3dedfba5d6cd487)
- Ubuntu 18 / ros melodic / gazebo9 from debian packages
- Debian package ros-melodic-ros-type-introspection is already the newest version (2.0.4-1bionic.20200530.015709)

Please let me know if this is a known issue.

Otherwise I'll be happy to provide more info / explore more simplified configurations to make it easier to reproduce. I'm not sure whether it's related to gazebo.

ljburtz, 2020-07-02:
update about issue B:
I could also reproduce it with the .bag from issue #93 
(after ~5 minutes in rosbag play loop mode)
The layout xml of issue #93 is from a version of Plotjuggler that's too old. So I used this one with random topics plotted in XY
[reproduceIssue93.zip](https://github.com/facontidavide/PlotJuggler/files/4862974/reproduceIssue93.zip)

ljburtz, 2020-07-02:
update about issue A :
It works great with the [AppImage] (https://github.com/facontidavide/PlotJuggler#download-and-run-ubuntu-1604-xenial-or-later)! No seg fault with the topics from gazebo or any that I tested so far.

So maybe some library dependency is skewed on my system?

facontidavide, 2020-07-02:
I will investigate

facontidavide, 2020-07-07:
I think I solved issue A. can you try the last version on master?

ljburtz, 2020-07-08:
Fantastic thanks for the fast fix. I've tested the latest version on master: no more issue. Happy to close the ticket!

closed 2020-07-08 by ljburtz