PlotJuggler/PlotJuggler #301: Segfault when using "Select topics to be published" window

SammyRamone, 2020-07-02:
I'm getting a segmentation fault when clicking on the "OK" button in the "Select topics to be published" window.
It happens for different rosbags. I can reproduce this with the `fake_joint_state.bag` example. It does not matter what I choose in the window. Closing it with without pressing "OK" works.
I'm using the current master under ROS melodic on an Ubuntu 18.04. I also tried the apt package (2.8.1), with the same results.
This is the terminal output that I get:

```
rosrun plotjuggler PlotJuggler   
"libbackward.so" :  "Failed to extract plugin meta data from '/home/marc/catkin_ws/devel/.private/plotjuggler/lib/plotjuggler/libbackward.so'"
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadROS.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamROS.so" : is a DataStreamer plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libRosoutPublisher.so" : is a StatePublisher plugin
"libRosTopicPublisher.so" : is a StatePublisher plugin
Gtk-Message: 15:23:02.976: GtkDialog mapped without a transient parent. This is discouraged.
The loading operation took 20 milliseconds
Stack trace (most recent call last):
#31   Object "/home/marc/catkin_ws/devel/.private/plotjuggler/lib/plotjuggler/PlotJuggler", at 0x562cfc8626d9, in _start
#30   Source "/build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c", line 310, in __libc_start_main [0x7fc03996fb96]
#29   Source "/home/marc/catkin_ws/src/PlotJuggler/plotter_gui/main.cpp", line 182, in main [0x562cfc861f51]
        180:     w.show();
        181:     splash.finish(&w);
      > 182:     return app.exec();
        183:   }
        184:   MainWindow w(parser);
        185:   w.show();
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc03a90f9b3, in QCoreApplication::exec()
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc03a906909, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc03a96188e, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#25   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fc037bda6db, in g_main_context_iteration
#24   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fc037bda64f, in g_main_context_dispatch
#23   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fc037bda416, in g_main_context_dispatch
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7fc030b1625f, in QXcbGlIntegrationPlugin::qt_metacall(QMetaObject::Call, int, void**)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7fc03aea32ea, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7fc03aecc054, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7fc03aeca582, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc03a9088d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b68e103, in QApplication::notify(QObject*, QEvent*)
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b68683b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#15   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b6e27d9, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b6dfea4, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#13   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b68d631, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool)
#12   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc03a9088d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#11   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b68e65e, in QApplication::notify(QObject*, QEvent*)
#10   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b68683b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#9    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b8096fa, in QMenu::event(QEvent*)
#8    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b6c5047, in QWidget::event(QEvent*)
#7    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b8073d2, in QMenu::mouseReleaseEvent(QMouseEvent*)
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b8065aa, in QMenu::leaveEvent(QEvent*)
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b7ff0fb, in QMenu::setIcon(QIcon const&)
#4    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b68280b, in QAction::activate(QAction::ActionEvent)
#3    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b680121, in QAction::triggered(bool)
#2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc03a93766e, in QMetaObject::activate(QObject*, int, int, void**)
#1    Source "/home/marc/catkin_ws/src/PlotJuggler/plugins/ROS/TopicPublisherROS/statepublisher_rostopic.cpp", line 150, in TopicPublisherROS::filterDialog(bool) [0x7fbffd2e960c]
        147:       }
        148:     }
        149: 
      > 150:     _publish_clock = dialog->ui()->radioButtonClock->isChecked();
        151: 
        152:     if (_enabled && _publish_clock)
        153:     {
#0    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc03b778400, in QAbstractButton::isChecked() const
Segmentation fault (Address not mapped to object [0x40000000d])
[1]    11090 segmentation fault (core dumped)  rosrun plotjuggler PlotJuggler
```

If you need any more information, I will gladly provide them

SammyRamone, 2020-07-02:
When trying it very often, it works a few times. I don't see any pattern. Maybe it is some sort of race condition?

facontidavide, 2020-07-07:
I just pushed a potential solution, or at least the only one I can think of, since I am unable to reproduce the error.

facontidavide, 2020-07-07:
FIXED

closed 2020-07-07 by facontidavide

SammyRamone, 2020-07-08:
Works fine now. Thanks for the fix :)