PlotJuggler/PlotJuggler #304: ROS2 colcon install

lucastliu, 2020-07-17:
Using master branch (also tried ros2 branch). How do I install this package from source (ROS2, eloquent)? I would like to use colcon, and not catkin. I clone the repository into my src directory, and attempt to run `colcon build --packages-select PlotJuggler`, (this is how I install all of my other packages) but this fails with message `WARNING:colcon.colcon_core.package_selection:ignoring unknown package 'PlotJuggler' in --packages-select
`

Please Advise

lucastliu, 2020-07-19:
I was missing some of the Qt5 dependencies. Additionally, I had to manually add the plotjugglers_msgs package dependency. The rpi is rather resource constrained, and runs out of virtual memory when attempting to compile.

Note: command for package should be `colcon build --packages-select plotjuggler` , with plotjuggler all lowercase, and not "PlotJuggler"

lucastliu, 2020-07-19:
Update: I have managed to successfully build. There is some missing code in the file ` PlotJuggler/plugins/ROS/CMakeLists.txt `
When using AMENT, it seems that the tf2_ros dependency was forgotten. This can be remedied by adding `find_package(tf2_ros REQUIRED)` to the list directly under `elseif(COMPILING_WITH_AMENT)` and adding the line `tf2_ros` to the list of `set(AMENT_DEPENDENCIES`

side note: not a functional error, but if someone is fixing this file anyways, the constant `AMENT_DEPENDENCIES` is misspelled as `AMENT_DEPENDECIES`

If you would like me to create a pull request, please let me know.

lucastliu, 2020-08-06:
In order to be able to run the plotjuggler with `ros2 run plotjuggler PlotJuggler`, an additional change must be made in `CMakeLists.txt`. Under the compiling with ament section, add the line `find_package(ament_cmake REQUIRED)`

The line near the bottom of `package.xml` must also be uncommented `<build_type condition="$ROS_VERSION == 2">ament_cmake</build_type>`

lucastliu, 2020-08-06:
@facontidavide again, let me know if you would like me to submit a pull request for these changes / how you would like to proceed.

facontidavide, 2020-08-06:
yes, I would really really appreciate a pull request :smile:

lucastliu, 2020-08-07:
Pull requests submitted. Note that the msgs dependency also needed a similar change if you need to build both from source for ROS2 eloquent.

lucastliu, 2020-08-10:
@soldierofhell @facontidavide bumping for visibility. Please let me know if the fixes worked for you / any changes needed for pull request approval.

closed 2020-08-11 by facontidavide

soldierofhell, 2020-08-27:
Sorry for late reply. As for me it works now, but not ROS2 branch, only master :)

PR #314 (2020-08-11): build fixes to work on ROS2 eloquent
Resolves #304 / #306 , allows PlotJuggler to be built from source and run on ROS2 eloquent.