PlotJuggler/PlotJuggler #305: Feature Request: Plot the trajectory for vector

gumpcarl, 2020-07-17:
Trajectory is import for path planning. But it is not easy to display via time-series. Hope the next version can surpport the single frame display of display.For example:
we have a vector variable, t.
struct Point{
   double x;
   double y;
   double z;
};
std::vector<Point> t{...}.
Plotjuggler can plot the t[0:end].x, t[0:end].y on the plot pannel.

Thank you!

facontidavide, 2020-07-17:
I hope that too. :stuck_out_tongue: 

It is a complicated problem really, because all the data structure in th code is based on the assumption that any plot is a timeseries.

Even if we write a plugin that can read these points and create a vector of them, a huge amount of changes are required.

aeudes, 2020-07-18:
Having more complex data representation will be awesome (especially for custom series) but in the mean time maybe a simpler solution exist here, no?
It's still possible to unroll the structure and make each field as a time serie. The difference from standard data, is that we switch from message converting to only one time value to one message generate multiple times values. If we consider that we keep only the last update value for message of this sort (we don't want to monitor change in trajectory but only last version of trajectory), it is possible to convert the message to multiple time series with arbitrary time index (or from the structure if given) to one time serie for each attributes ?
traj/Point[] => traj/x , traj/y, traj/z.
It could be very useful to display [path](http://docs.ros.org/melodic/api/nav_msgs/html/msg/Path.html) or [trajectory](http://docs.ros.org/melodic/api/trajectory_msgs/html/msg/JointTrajectory.html) for example.

facontidavide, 2020-07-18:
I am planning to do this in PlotJuggler 3.0 (currently under development). 

As I said, it requires a  massive refactoring. In other words: I want do it and I will do it, but this will take still few weeks (or months).

To anybody interested:

**If you want to speed up the development of PlotJuggler 3, consider becoming a [Github Sponsor](https://github.com/sponsors/facontidavide)**

roger-, 2021-06-18:
Looking forward to this, is it still planned? 3.2 doesn't seem to have it.

facontidavide, 2021-06-18:
Not happening, sorry

facontidavide, 2022-01-29:
rejoice: https://slides.com/davidefaconti/plotjuggler-reactive-scripts/fullscreen

closed 2022-01-29 by facontidavide

salmagro, 2022-02-04:
Thank you very much, @facontidavide. It was a needed tool.