PlotJuggler/PlotJuggler #306: ROS2 Eloquent ROS Plugins Non-Functional

lucastliu, 2020-07-19:
After installing PlotJuggler, I can manually run the program from the PlotJuggler executable, but none of the plugins under the ROS folder appear functional (i.e. DataStreamROS2)

Messages in terminal on startup:

` "libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadROS2.so" :  "Cannot load library .../dev_ws/install/plotjuggler/lib/plotjuggler/libDataLoadROS2.so: (librosbag2_transport.so: cannot open shared object file: No such file or directory)"
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamROS2.so" :  "Cannot load library .../dev_ws/install/plotjuggler/lib/plotjuggler/libDataStreamROS2.so: (librosbag2_transport.so: cannot open shared object file: No such file or directory)"
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libTopicPublisherROS2.so" :  "Cannot load library .../dev_ws/install/plotjuggler/lib/plotjuggler/libTopicPublisherROS2.so: (librosbag2_transport.so: cannot open shared object file: No such file or directory)"

`

The GUI/program launches, but no plugin functionalities are available. rosbag2_transport was also installed and the file in question can be found at ...ros2_eloquent/install/rosbag2_transport/lib/librosbag2_transport.so

lucastliu, 2020-07-19:
Is this issue related to the fact that I cannot run PlotJuggler using ros2 run? (ros 2 equivalent of rosrun).

I can only run by manually running the executable. The command (which works for all of my other packages) `ros2 run plotjuggler PlotJuggler` simply results in the message `Package 'plotjuggler' not found
`

I have sourced the underlay ros2_eloquent, as well as the overlay personal workspace dev_ws (where the plotjuggler package is installed from source)

lucastliu, 2020-07-20:
Issues seems to resolve if one launches the nodes first, and plotjuggler second.

facontidavide, 2020-07-20:
interesting, thanks for reporting.

I need to investigate

soldierofhell, 2020-08-06:
Hi @lucastliu 

> Issues seems to resolve if one launches the nodes first, and plotjuggler second.

Could you please clarify what do you mean by "launching nodes first"? I've got the same symptoms on eloquent, e.g. there's no plotjuggler in `ros2 pkg list` and hence `ros2 run plotjuggler PlotJuggler` returns "not found".

I'm new to plotjuggler and don't know it's internal architecture

Regards,

lucastliu, 2020-08-06:
As far as "launching nodes first," that simply meant that I ran `ros2 run [insert package] [insert node]` for my other ROS2 project components before I ran ros2 run plotjuggler PlotJuggler` . I think I had conflated multiple issues, and that this is not actually the problem. I will edit my earlier answer to correct the mistake.

The first thing to check is whether you are in the correct directory, both for package installation and for running the program. This means the directory you run `colcon build` from needs to be the same directory you run `ros2 run plotjuggler PlotJuggler`

The second thing to check is to make sure you have sourced the underlay / overlay before running as described [here](https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/#source-ros-2-environment)

Third, read the build output carefully, and ensure the package is actually successfully built. I had a lot of trouble here (I built the plotjuggler package from source). 

If you are installing with colcon, I would look at my other issue #304 , as I am guessing you may run into similar issues if you have build failures.

lucastliu, 2020-08-06:
I think this issue is better addressed in #304 . The original problem in this issue is somewhat tied to #304 , so I'll close this issue. @soldierofhell please see #304 for additional details, and fixes.

closed 2020-08-06 by lucastliu