PlotJuggler/PlotJuggler #307: Question: Plans to support ros2-foxy?

alexselwa, 2020-07-21:
Super excited to find PlotJuggler!! 

Apologies for asking this in a GitHub issue but I was curious if anyone is working on support for newer ROS2 distros like Foxy? On first glance there were a few changes that broke the ros2 branch, such as `rmw_message_info_t` to `rclcpp::MessageInfo` (see `subscription_base.hpp` in rclcpp). 

 Thanks for developing this awesome tool!

facontidavide, 2020-07-23:
That will surely require some work and right now i am focused on other tasks. Any contribution from the community is welcome.

facontidavide, 2020-09-16:
this is done in theory. Waiting to check #328 before closing this issue too.

facontidavide, 2020-09-17:
done 

[image]

closed 2020-09-17 by facontidavide

alexselwa, 2020-09-21:
Amazing, thank you!!