PlotJuggler/PlotJuggler #308: [Build Error] 3rdparty qwt building for ROS-noetic on Arch Linux fails

acxz, 2020-07-23:
Thanks for contributing to PlotJuggler. You are great!

So are you for creating PlotJuggler!

# If you ignore these instructions, I will ignore your issue

Please allow me to better understand your problem providing the following information:

- Please use always the master branch or the latest tag, if you can.
Latest tag 2.8.3 is used. Doubt switching to latest head on master will help since this a vendored library issue.

- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".

Not applicable

- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

Not applicable

[image]

Not applicable

When attempting to build this package with `catkin build`:
I am running into the following error:
```
[  7%] Building CXX object 3rdparty/qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_renderer.cpp.o
In file included from /home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp:13:
/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_layout.h:84:51: warning: ‘constexpr QFlags<T>::QFlags(QFlags<T>::Zero) [with Enum = QwtPlotLayout::Option; QFlags<T>::Zero = int QFlags<QwtPlotLayout::Option>::Private::*]’ is deprecated: Use default constructor instead [-Wdeprecated-declarations]
   84 |         const QRectF &plotRect, Options options = 0x00 );
      |                                                   ^~~~
In file included from /usr/include/qt/QtCore/qglobal.h:1302,
                 from /home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_global.h:13,
                 from /home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.h:13,
                 from /home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp:10:
/usr/include/qt/QtCore/qflags.h:123:80: note: declared here
  123 | ATED_X("Use default constructor instead") Q_DECL_CONSTEXPR inline QFlags(Zero) noexcept : i(0) {}
      |                                                                   ^~~~~~

/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp:94:53: error: return type ‘class QPainterPath’ is incomplete
   94 |     const QWidget* canvas, const QRectF &canvasRect )
      |                                                     ^
/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp: In function ‘void qwtCanvasClip(const QWidget*, const QRectF&)’:
/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp:107:18: error: aggregate ‘QPainterPath clipPath’ has incomplete type and cannot be defined
  107 |     QPainterPath clipPath;
      |                  ^~~~~~~~
/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp: In member function ‘void QwtPlotRenderer::renderDocument(QwtPlot*, const QString&, const QString&, const QSizeF&, int)’:
/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp:312:41: warning: ‘virtual void QPdfWriter::setPageSizeMM(const QSizeF&)’ is deprecated: Use setPageSize(QPageSize(size, QPageSize::Millimeter)) instead [-Wdeprecated-declarations]
  312 |         pdfWriter.setPageSizeMM( sizeMM );
      |                                         ^
In file included from /home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp:71:
/usr/include/qt/QtGui/qpdfwriter.h:98:10: note: declared here
   98 |     void setPageSizeMM(const QSizeF &size) override;
      |          ^~~~~~~~~~~~~
/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp: In member function ‘virtual void QwtPlotRenderer::render(QwtPlot*, QPainter*, const QRectF&) const’:
/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp:507:64: warning: ‘void QWidget::getContentsMargins(int*, int*, int*, int*) const’ is deprecated: use contentsMargins() [-Wdeprecated-declarations]
  507 |         plot->getContentsMargins( &left, &top, &right, &bottom );
      |                                                                ^
In file included from /usr/include/qt/QtWidgets/qframe.h:44,
                 from /home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot.h:16,
                 from /home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp:11:
/usr/include/qt/QtWidgets/qwidget.h:530:10: note: declared here
  530 |     void getContentsMargins(int *left, int *top, int *right, int *bottom) const;
      |          ^~~~~~~~~~~~~~~~~~
/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp: In member function ‘virtual void QwtPlotRenderer::renderCanvas(const QwtPlot*, QPainter*, const QRectF&, const QwtScaleMap*) const’:
/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp:859:22: error: aggregate ‘QPainterPath clipPath’ has incomplete type and cannot be defined
  859 |         QPainterPath clipPath;
      |                      ^~~~~~~~
/home/acxz/vcs/git/github/ros-noetic-arch/ros-noetic-plotjuggler/src/PlotJuggler-2.8.3/3rdparty/qwt/src/qwt_plot_renderer.cpp:883:22: error: aggregate ‘QPainterPath clipPath’ has incomplete type and cannot be defined
  883 |         QPainterPath clipPath;
      |                      ^~~~~~~~
make[2]: *** [3rdparty/qwt/src/CMakeFiles/qwt_static.dir/build.make:212: 3rdparty/qwt/src/CMakeFiles/qwt_static.dir/qwt_plot_renderer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2507: 3rdparty/qwt/src/CMakeFiles/qwt_static.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```

Sys info:
OS: Arch Linux
ROS: Noetic
GCC: 10.1.0

Would there be anyway for me to compile against Arch Linux's packaged qwt, instead of using the one vendored inside of plotjuggler?

Thanks!

facontidavide, 2020-08-07:
> Would there be anyway for me to compile against Arch Linux's packaged qwt, instead of using the one vendored inside of plotjuggler?

no. I made specific changes to qwt and PlotJuggler MUST use its own version.

I have the feeling that for some reason it is mixing header files from the installed Qwt package and the one in PJ...

Not having Arch linux, this might take a while for me to address.

acxz, 2020-08-07:
Thanks for responding!

> no. I made specific changes to qwt and PlotJuggler MUST use its own version.

I see, kk if I get time I'll try to PR some changes to your version of qwt to make it Arch compatible.

> I have the feeling that for some reason it is mixing header files from the installed Qwt package and the one in PJ...

I don't think this is happening because of the paths used in the error logs. I think the qwt version you based your qwt on, was tested against an older version of QT that Arch does not have. This means we would need to change your qwt to work with the newer QT or rebase it your qwt changes off of a newer qwt version.

It's weird tho that you have other 3rdparty libs that you updated to QT 5.15 so I kinda assumed this one would also be updated for the latest QT version.

facontidavide, 2020-08-07:
I have just compiled the latest PJ using Qt 5.14.2 installed in my computer and there is no error.

acxz, 2020-08-07:
Thanks for testing that, so I guess we can pin down the issue due to a newer QT 5.15 install.

acxz, 2020-08-07:
Okay so I figured out the issue, there are two ways to go about this.

One is the solution I propose in the PR, which is the easiest to incorporate. 

The other more principled way is to fork qwt and apply plotjuggler specific changes to it. We then include this forked qwt in the plotjuggler repo as a submodule or a git subtree. 
This allows us to track changes in upstream qwt and we can rebase plotjuggler specific commits on top of it easily when we need to update qwt.

I think for now we can merge my proposed PR and then eventually work towards creating this qwt fork/submodule.

PR #313 (2020-08-07): add qtpainterpath.h
Resolves #308
closed 2020-08-07 by facontidavide