PlotJuggler/PlotJuggler #309: DataStreamROS2 Plugin incompatible subscription

lucastliu, 2020-07-24:
I am seeing some of my topics show up under the list for streaming ROS2 Topic Subscriber in plotjuggler. However, after highlighting and clicking Add, they never show up. Topics have several nodes that are publishing and receiving.

I am noticing that the only topics that have this issue are ones with a Quality of Service (qos) profile (in my case, qos_profile_sensor_data), while topics with default publishers do not have this issue.

I believe this may be because the DataStreamROS2 plugin is creating an incompatible subscription. 

See the Section QoS Compatibilities on this [QoS page](https://index.ros.org//doc/ros2/Concepts/About-Quality-of-Service-Settings/#qos-compatibilities)

Many subscriber types may not be compatible with a sensor_data QoS, therefore causing the plotjuggler plugin to fail for these topics.

Any way to change the way a subscription generated inside the DataStreamROS2 plugin?

lucastliu, 2020-07-24:
I've located the source of the issue - problem is QoS compatibility as outlined above. Unclear how this should be dealt with on the master branch, but here's what happens:

In [generic_subscription.hpp](https://github.com/facontidavide/PlotJuggler/blob/master/plugins/ROS/ros2_parsers/generic_subscription.hpp)

subscription constructor sets qos profile on line 105. A subscription made using rcl_subscription_get_default_options() will not be able to receive data from certain QoS publishers. Change the constructor to use a QoS suitable to your needs.

Example: I have publishers that operate under the sensor data QoS. Default subscription will fail. So I change my QoS options as follows:

`
  rcl_subscription_options_t options = rcl_subscription_get_default_options();
  options.qos.reliability = RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT;
  options.qos.durability = RMW_QOS_POLICY_DURABILITY_VOLATILE;
  
  return options;
`

Now when i rebuild PlotJuggler and run, I am able to see topics that have restrictive QoS publishers.

facontidavide, 2020-08-29:
This is an issue that apparently is being addressed in rosbag2 in Foxy. I have to have a look at it.

But do you think your change is sufficient to cover all the use cases? In that case, would you submit a PR?

lucastliu, 2020-08-29:
Here is the relevant documentation page: [ROS 2QoS](https://index.ros.org/doc/ros2/Concepts/About-Quality-of-Service-Settings/)

I can definitely submit a PR, but you should be aware of a potential downside. I can submit a PR with a subscription profile that will be able to receive information from any publishing source (least restrictive QoS). This will prevent data from simply not appearing due to incompatible QoS. 

However, this does mean that the subscription for PlotJuggler will be willing to accept "low quality" data connections. If the user expects a certain QoS in their data, it is now on them to ensure that their publishers have the appropriate QoS settings.  

To be more concrete (refer to documentation page for full details), the profile would have: 
```
reliability = best effort
durability = volatile
deadline = Default
liveliness = automatic
lease duration = Default
```

Conclusion: The fix should allow all topics to be received by PlotJuggler subscription (not true on current master branch). However, if users expect a certain level of QoS, it will be on them to ensure that their Publisher has an appropriate QoS profile.

I think this is an acceptable change, but please let me know how you would like to proceed @facontidavide

facontidavide, 2020-08-29:
I need to take some time and see how people in rosbag2 addressed this.

soldierofhell, 2020-11-19:
@facontidavide, that's really a blocker for many analysis, e.g. all the sensors data from gazebo. Maybe expose those options in gui like in rviz2?
[image]

facontidavide, 2020-11-19:
I will **eventually** work on this.

I will add a kind reminder that the best way to ensure that I work on PlotJuggler is to become a [Github Sponsor](https://github.com/sponsors/facontidavide).

Alternatively you may address this yourself and send a Pull Request :smile:

soldierofhell, 2020-11-19:
Ok, got it :) I think short term most important is just to make a decision what to do to make it funcional by default. I would suggest to:
1. release constraints as suggested by @lucastliu 
2. make change in gui to make it more flexible

1 is straightforward, 2 hopefully someone will find time to prepare

facontidavide, 2020-11-23:
PlotJuggler 3 should fix this (I implemented solution 1)

closed 2021-01-23 by facontidavide