PlotJuggler/PlotJuggler #310: Segmentation fault (Signal sent by the kernel [(nil)])

nag92, 2020-08-01:
Plot juggler(2.8.1) keeps crashing on me on it seems seems similar to #216.  I am running 18.04.  
I get the following error output:

```bash
Stack trace (most recent call last):
#20   Object "", at 0xffffffffffffffff, in 
#19   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x562795b7e6d9, in main
#18   Source "/build/glibc-2ORdQG/glibc-2.27/csu/../csu/libc-start.c", line 310, in __libc_start_main [0x7f8f87226b96]
#17   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x562795b7df51, in main
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f8f881c69b3, in QCoreApplication::exec()
#15   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f8f881bd909, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#14   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f8f8821888e, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#13   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f8f854916db, in g_main_context_iteration
#12   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f8f8549164f, in g_main_context_dispatch
#11   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f8f85491416, in g_main_context_dispatch
#10   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f8f88218528, in QEventDispatcherGlib::flush()
#9    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f8f88217d2d, in QTimerInfoList::activateTimers()
#8    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f8f881bf8d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#7    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f8f88f45103, in QApplication::notify(QObject*, QEvent*)
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f8f88f3d83b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f8f881ef07a, in QObject::event(QEvent*)
#4    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f8f881fb327, in QTimer::timerEvent(QTimerEvent*)
#3    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f8f881fafc6, in QTimer::timeout(QTimer::QPrivateSignal)
#2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f8f881ee66e, in QMetaObject::activate(QObject*, int, int, void**)
#1    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x562795b97146, in main
#0    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x562795bb28ca, in main
Segmentation fault (Signal sent by the kernel [(nil)])
Segmentation fault (core dumped)

```

I have attached my bag file from when I got this issue but I saw this issue while running other program has well. 

[2020-08-01-12-11-46.zip](https://github.com/facontidavide/PlotJuggler/files/5010906/2020-08-01-12-11-46.zip)

facontidavide, 2020-08-01:
I have just opened the file directly with PJ 2.8.1 and there is no error.

Are you playing this with **rosbag play**  and connecting to the live data with the streamer?

If that is the case, the error has been fixed already in newer versions.

nag92, 2020-08-02:
building from source seems to have resolved the issue.

closed 2020-08-02 by nag92