PlotJuggler/PlotJuggler #315: Doesn't it support Windows now?

Chunting, 2020-08-10:
Hi,
It seems that PlotJuggler does not support windows any more, doesn't it?

 I'm working on an old version which could run on windows, just double check if I can upgrade to the newest one.

Thanks,

Chunting

facontidavide, 2020-08-10:
I need to maintain the windows version. It might compile.. Or not.

Too little time to deal with everything 😕

seanmcleod, 2020-08-10:
Commit https://github.com/facontidavide/PlotJuggler/commit/c33d138fd5dda090b7f1cafaee53660ec56fedbd 3 days ago failed to build the Windows version on AppVeyor and the Windows Github action.

The AppVeyor build - https://ci.appveyor.com/project/facontidavide59577/plotjuggler/builds/34546361 fails during compilation.

```sh
[ 76%] Building CXX object plotter_gui/CMakeFiles/PlotJuggler.dir/PlotJuggler_autogen/mocs_compilation.cpp.obj
In file included from C:\projects\plotjuggler\build\plotter_gui\PlotJuggler_autogen\CQ57MUOM5H/moc_datastreamer_base.cpp:9:0,
                 from C:\projects\plotjuggler\build\plotter_gui\PlotJuggler_autogen\mocs_compilation.cpp:2:
c:\projects\plotjuggler\include\plotjuggler\datastreamer_base.h:29:8: error: 'mutex' in namespace 'std' does not name a type
   std::mutex& mutex()
        ^
```

The Github Action - https://github.com/facontidavide/PlotJuggler/runs/960006501 fails earlier during configuration.

```sh
CMake Error at D:/a/PlotJuggler/Qt/5.12.7/mingw73_64/lib/cmake/Qt5Core/Qt5CoreMacros.cmake:211 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  D:/a/PlotJuggler/Qt/5.12.7/mingw73_64/lib/cmake/Qt5Core/Qt5CoreMacros.cmake:281 (_QT5_PARSE_QRC_FILE)
  plotter_gui/CMakeLists.txt:4 (QT5_ADD_RESOURCES)

-- Configuring incomplete, errors occurred!
See also "D:/a/PlotJuggler/PlotJuggler/build/CMakeFiles/CMakeOutput.log".

##[error]Process completed with exit code 1.
```

maroc81, 2020-08-11:
I was able to produce a windows build using msys2.  The general process is

1. Download and install msys2 according to instructions at https://www.msys2.org/
2. Run the msys2 64 bit termnial (MSYS2 MinGW 64)
3. Install qt packages with 
```
pacman -S mingw-w64-x86_64-qt5 mingw-w64-x86_64-qt5-static mingw-w64-x86_64-gdb mingw-w64-x86_64-clang mingw-w64-x86_64-qt-creator
```
4. Install cmake and make
```
pacman -S cmake make
```
5. Clone repo and create build directory under project root
6. Run cmake command followed by make
```
cmake -G "MSYS Makefiles" ..
make
```

After building, you should be able to run the exe with the command "./bin/PlotJuggler.exe".  However, you can't yet run the exe outside of the msys2 command line.  In order to do that you have to do the following:

1. Edit cmake_install.cmake and replace the C:/Program Files path with "../install" so the install will place files in PlotJuggler/install
2. Run cmake command
```
cmake --install ./ --prefix ../install
```
3. Run windeployqt to copy all dependencies
```
windeployqt.ext --release --compiler-runtime ../install/bin/PlotJuggler.exe
```
4. At this point, a few dependencies must be copied by hand with the commands
```
cp /c/msys64/mingw64/bin/libicuuc65.dll ../install/bin
cp /c/msys64/mingw64/bin/libpcre2-16-0.dll ../install/bin
cp /c/msys64/mingw64/bin/libzstd.dll ../install/bin
cp /c/msys64/mingw64/bin/libicudt65.dll ../install/bin
cp /c/msys64/mingw64/bin/zlib1.dll ../install/bin
cp /c/msys64/mingw64/bin/libdouble-conversion.dll ../install/bin
cp /c/msys64/mingw64/bin/libicuin65.dll ../install/bin
cp /c/msys64/mingw64/bin/libharfbuzz-0.dll ../install/bin
cp /c/msys64/mingw64/bin/libpng16-16.dll ../install/bin
cp /c/msys64/mingw64/bin/libglib-2.0-0.dll ../install/bin
cp /c/msys64/mingw64/bin/libfreetype-6.dll ../install/bin
cp /c/msys64/mingw64/bin/libgraphite2.dll ../install/bin
cp /c/msys64/mingw64/bin/libintl-8.dll ../install/bin
cp /c/msys64/mingw64/bin/libbz2-1.dll ../install/bin
cp /c/msys64/mingw64/bin/libpcre-1.dll ../install/bin
cp /c/msys64/mingw64/bin/libiconv-2.dll ../install/bin
cp /c/msys64/mingw64/bin/libssl-1_1-x64.dll ../install/bin
cp /c/msys64/mingw64/bin/libcrypto-1_1-x64.dll ../install/bin
cp /c/msys64/mingw64/bin/libjpeg-8.dll ../install/bin
cp /c/msys64/mingw64/bin/libjasper-4.dll ../install/bin
```

You can now zip up the install directory and extract to desired location.  If you want to reduce the size of the files, run the command

```
find ../install/bin/ --type f -exec strip {} \;
```
Before zipping

seanmcleod, 2020-08-11:
Thanks @maroc81. Does `windeployqt.exe` not pick up all the required dependencies and so that's the reason for the list that has to be copied by hand?

Actually I guess based on the name I guess `windeployqt.exe` only copies the QT DLL dependencies and not the non-QT DLL dependencies?

How large is the zipped up install directory?

maroc81, 2020-08-11:
That's right.  `windeployqt.exe` does not pick up all of the dependencies.  I think some of the libraries that Qt depends on are the ones it misses.  There may be some option to pass so it finds them and includes them but I didn't see any obvious option to do that.

After zipping it's around 43MB, extracted it's around 98MB.

seanmcleod, 2020-08-12:
@maroc81 any chance of you being able to upload the zip file somewhere for download? I'd like to test out PlotJuggler on WIndows a bit first before diving in to build it locally etc.

maroc81, 2020-08-12:
I created a release from my forked version.

https://github.com/maroc81/PlotJuggler/releases/tag/2.8.3_windows

It has some unmerged changes I made for showing a warning message box if the CSV has lines that are skipped.

lukegluke, 2020-08-12:
Hi @maroc81,

Thanks for your work. Trying to run this, but got in console `Failed to extract plugin meta data from` for many libraries: libbz2-1.dll, libcryptro-1_1-x64, libglib-2.0-0.dll etc. and also for Qt5Core.dll, Qt5Gui.dll.
Then on clicking load data menu in console once got:
`Exception code=0xc0000005 flags=0x0 at 0x00007FF806E9CB96. Access violation - attempting to read data at address 0x000000000DE80000`
And finally `Ulog: error loading definitions` on file opening.

seanmcleod, 2020-08-12:
@lukegluke I see the same messages about failing to extract meta data but I'm able to use the Load data menu option to load a .csv file.

maroc81, 2020-08-12:
@lukegluke I get the same "failed to extract" errors which are due to PlotJuggler trying to load all the dlls as plugins.  However, PlotJuggler eventually loads and runs fine.  I tested on Windows 7 and Windows 10.  What windows version are you running? 

It sounds like I have some windows library loaded (due to this being a development system) that you are missing.  Let me see if I can figure that out.

lukegluke, 2020-08-12:
> @lukegluke I see the same messages about failing to extract meta data but I'm able to use the Load data menu option to load a .csv file.

Thanks, forgot about csv files - they are working, yes.

maroc81, 2020-08-12:
So its a problem loading ulog files then?  I did not test loading ulog files.  If you can provide one for testing or can point me to one I can download I can see if it will work for me.

lukegluke, 2020-08-12:
> What windows version are you running?

Windows 10 Pro x64.

> So its a problem loading ulog files then? I did not test loading ulog files. If you can provide one for testing or can point me to one I can download I can see if it will work for me.

I think so. You can download any publicly available ulog from PX4 flight review https://logs.px4.io/browse (choose one and click download log button on top)
For example: https://logs.px4.io/download?log=a0291e2c-2108-4672-a0b4-63beec1770bc

facontidavide, 2020-08-13:
Thanks for everybody taking the initiative here.

Rejoice: https://github.com/facontidavide/PlotJuggler/releases/download/2.8.3/PlotJuggler_WIN_Installer-2.8.3.exe

lukegluke, 2020-08-13:
Hi @facontidavide,

Thanks for work! But there is the same issue with your version as with @maroc81 - csv works, but ulogs failed to open.

maroc81, 2020-08-13:
@lukegluke I have confirmed it does not work with my windows build. I can however load the ulog with a linux build of PlotJuggler.  I thought it might be a problem with the qt version I used to do the windows build but have not confirmed yet.

facontidavide, 2020-08-14:
It is really weird that the ULOG plugins works in Linux but not inWindows. I am investigating

facontidavide, 2020-08-14:
Problem with ULOG fixed (only windows was affected. I will release version 2.8.4 soon

closed 2020-08-14 by facontidavide