PlotJuggler/PlotJuggler #318: Segmentation fault in some rare cases

AlexisTM, 2020-08-14:
**Version**: Last Ubuntu 18.04 melodic release: Version: 2.8.3-1bionic.20200801.210347

**Description**
I got a segmentation fault when loading a layout but **I cannot reproduce it**.

**Steps I took**
* Load data
* Create a layout [Multiple tabs, both x-time and xy graphs with multiple graphs per page] 
* Save the layout
* Modify the layout
* Load the layout

```css

#31   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6baf383b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#30   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6bc766fa, in QMenu::event(QEvent*)
#29   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6bb32047, in QWidget::event(QEvent*)
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6bc74275, in QMenu::mouseReleaseEvent(QMouseEvent*)
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6bc7401f, in QMenu::leaveEvent(QEvent*)
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3e6ad758d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6bafb65e, in QApplication::notify(QObject*, QEvent*)
#24   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6baf383b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6bc766fa, in QMenu::event(QEvent*)
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6bb32047, in QWidget::event(QEvent*)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6bc743d2, in QMenu::mouseReleaseEvent(QMouseEvent*)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6bc735aa, in QMenu::leaveEvent(QEvent*)
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6bc6c0fb, in QMenu::setIcon(QIcon const&)
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6baef80b, in QAction::activate(QAction::ActionEvent)
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3e6baed121, in QAction::triggered(bool)
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3e6ada4618, in QMetaObject::activate(QObject*, int, int, void**)
#15   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a3349a48, in main
#14   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a3348053, in main
#13   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a32f0153, in main
#12   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a32ececf, in main
#11   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a32e9aa8, in main
#10   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a331f3ce, in main
#9    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a33141b6, in main
#8    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a330d4a9, in main
#7    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a3306ed2, in main
#6    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a33069b3, in main
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3e6ada466e, in QMetaObject::activate(QObject*, int, int, void**)
#4    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a32e11be, in main
#3    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a32dd8df, in main
#2    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a331f070, in main
#1    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a33131ba, in main
#0    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55f2a3304f1d, in main
Segmentation fault (Address not mapped to object [0x58])
Segmentation fault
```

Note: Rename the files to .xml and .bag, they were renamed to be sharable on Github directly.

[gps_review(fix_inertialsense_rev_latlon).xml.txt](https://github.com/facontidavide/PlotJuggler/files/5073385/gps_review.fix_inertialsense_rev_latlon.xml.txt)

[turnaround_inplace.bag.txt](https://github.com/facontidavide/PlotJuggler/files/5073387/turnaround_inplace.bag.txt)

facontidavide, 2020-08-14:
thanks for reporting, but I can not reproduce it either :(

It is a shame that the backtrace misses debug information, I should check why

facontidavide, 2020-08-14:
I will send you an AppImage with debug information. You may use that and if that happens again, you may send me the extended backtrace.

AlexisTM, 2020-08-14:
Indeed, such a weak backtrace doesn't help much.

facontidavide, 2020-08-14:
Try using this, I believe it should contain the debug information. 

https://github.com/facontidavide/PlotJuggler/releases/download/2.8.3/PlotJuggler-Linux-ROS-2.8.3.AppImage

AlexisTM, 2020-08-14:
@facontidavide Yey! Faster than expected, adding tabs & removing tabs helped to reproduce it! 

```css
#31   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Gui.so.5", at 0x7fc8c1eaa4e0, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#30   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Core.so.5", at 0x7fc8c1b3426a, in QCoreApplication::notifyInternal(QObject*, QEvent*)
#29   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c24b0515, in QApplication::notify(QObject*, QEvent*)
#28   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c24ab05b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#27   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c250ab7a, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#26   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c250891c, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#25   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c24afb31, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool)
#24   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Core.so.5", at 0x7fc8c1b3426a, in QCoreApplication::notifyInternal(QObject*, QEvent*)
#23   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c24b0c18, in QApplication::notify(QObject*, QEvent*)
#22   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c24ab05b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#21   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c2630ab2, in QMenu::event(QEvent*)
#20   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c24edfc7, in QWidget::event(QEvent*)
#19   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c263005f, in QMenu::mouseReleaseEvent(QMouseEvent*)
#18   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c262c0eb, in QMenu::internalDelayedPopup()
#17   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c2625e51, in QMenu::setIcon(QIcon const&)
#16   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c24a3897, in QAction::activate(QAction::ActionEvent)
#15   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Widgets.so.5", at 0x7fc8c24a1411, in QAction::triggered(bool)
#14   Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Core.so.5", at 0x7fc8c1b62a8e, in QMetaObject::activate(QObject*, int, int, void**)
#13 | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject_impl.h", line 192, in impl
    | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h", line 552, in call<QtPrivate::List<>, void>
    | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h", line 495, in call
      Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 863, in operator() [0x45f64d]
#12   Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 1590, in MainWindow::loadLayoutFromFile(QString) [0x45c70b]
#11   Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 758, in MainWindow::xmlLoadState(QDomDocument) [0x458d01]
#10   Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/tabbedplotwidget.cpp", line 156, in TabbedPlotWidget::xmlLoadState(QDomElement&) [0x49248c]
#9    Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/plotmatrix.cpp", line 329, in PlotMatrix::xmlLoadState(QDomElement&) [0x486b27]
#8    Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/plotwidget.cpp", line 839, in PlotWidget::xmlLoadState(QDomElement&) [0x47f263]
#7    Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/plotwidget.cpp", line 1607, in PlotWidget::convertToXY() [0x478922]
#6    Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/plotwidget.cpp", line 1532, in PlotWidget::zoomOut(bool) [0x478002]
#5    Object "/tmp/.mount_PlotJuyhJzrR/usr/lib/libQt5Core.so.5", at 0x7fc8c1b62a8e, in QMetaObject::activate(QObject*, int, int, void**)
#4    Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 215, in MainWindow::onUndoableChange() [0x44ec2a]
#3    Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/mainwindow.cpp", line 630, in MainWindow::xmlSaveState() const [0x44bd86]
#2    Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/tabbedplotwidget.cpp", line 126, in TabbedPlotWidget::xmlSaveState(QDomDocument&) const [0x4920b9]
#1    Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/plotmatrix.cpp", line 278, in PlotMatrix::xmlSaveState(QDomDocument&) const [0x485a79]
#0    Source "/home/davide/ws_pj/src/PlotJuggler/plotter_gui/plotwidget.cpp", line 808, in PlotWidget::xmlSaveState(QDomDocument&) const [0x476438]
Segmentation fault (Address not mapped to object [0x58])
Segmentation fault
```

facontidavide, 2020-08-14:
good, I was able to reproduce it. Fix coming soon. Thanks again

facontidavide, 2020-08-14:
fixed! that was really a critical bug, thanks for taking the time to report it and describe it

[image]

closed 2020-08-14 by facontidavide

AlexisTM, 2020-08-14:
Everywhere I go, I just need to touch it to break it! Brilliant! :laughing: