PlotJuggler/PlotJuggler #322: couldnt update installation from source

ljburtz, 2020-08-15:
Hello, I wanted to upgrade from version 2.8.2 to 2.8.4
After pulling from the repo, I run `catkin build`

But I get this error that ` package `plotjuggler` because it has an unsupported package build type: `a
ment_cmake``

I must be missing something obvious. I've been using using the built from source version since the beginning.  Has the build type changed in the latest version?

Thanks! 

Louis

```
catkin build                                     
--------------------------------------------------------------------------                          
Profile:                     default                                                                
Extending:          [cached] /opt/ros/melodic                                                       
Workspace:                   /home/ubuntu/utilities/ws_plotjuggler                                  
--------------------------------------------------------------------------                          
Build Space:        [exists] /home/ubuntu/utilities/ws_plotjuggler/build                            
Devel Space:        [exists] /home/ubuntu/utilities/ws_plotjuggler/devel                            
Install Space:      [unused] /home/ubuntu/utilities/ws_plotjuggler/install                          
Log Space:          [exists] /home/ubuntu/utilities/ws_plotjuggler/logs                             
Source Space:       [exists] /home/ubuntu/utilities/ws_plotjuggler/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.
[build] Package table is up to date.
[build] Warning: Skipping package `plotjuggler` because it has an unsupported package build type: `a
ment_cmake`
[build] Note: Available build types:
[build]  - `catkin`
[build]  - `cmake`
[build] Summary: All 0 packages succeeded!
[build]   Ignored:   1 packages were skipped or are blacklisted.
[build]   Warnings:  None.
[build]   Abandoned: None.
[build]   Failed:    None.
[build] Runtime: 0.0 seconds total.
```

ljburtz, 2020-08-15:
I've checked https://github.com/facontidavide/PlotJuggler/blob/master/package.xml
and tested that 
```
echo $ROS_VERSION
1
```

ljburtz, 2020-08-15:
update: by changing this
```
 git diff package.xml
diff --git a/package.xml b/package.xml
index c1f8762..d188921 100644
--- a/package.xml
+++ b/package.xml
@@ -47,6 +47,7 @@
 
     <!-- The export tag contains other, unspecified, tags -->
     <export>
+        <build_type condition="$ROS_VERSION == 1">catkin</build_type>
         <build_type condition="$ROS_VERSION == 2">ament_cmake</build_type>
     </export>
 </package>
```

I can build successfully. but am I missing something else?

facontidavide, 2020-08-15:
catkin tools has a bug that I have reported. Compilation works with catkin make.

Good news is that you found the solution to my problem, thanks ;)

I will include this change

closed 2020-08-15 by facontidavide

ljburtz, 2020-08-15:
awesome thanks for the quick reply!