PlotJuggler/PlotJuggler #323: [feature request] quaternion to roll/pitch/yaw_deg from gazebo model states too!

ljburtz, 2020-08-15:
the quaternion to roll/pitch/yaw in degrees on the fly conversion is absolutely fantastic.
It workds great with IMU and my Pose messages too eg
[image]

However it does not work with the pose from gazebo's `model states` topic
[image]

Is there a simple fix for this feature?
Thanks a lot (I have tons of gazebo quaternions would be a life saver not to make custom time series for each!)

facontidavide, 2020-08-29:
The solution is potentially easy but would require PJ to depend on these messages.

Unfortunately, that will NOT happen. 

The not so easy solution is to inspect all the messages containing a quaternion and add RPY to it.
That might be done in the future.... but I am currently focusing on other features, so this is not one of my priorities.

**NOTE: if you or anyone want to influence the prioritization of features, consider becoming a sponsor.**

ljburtz, 2020-08-29:
Understood and fair point. I wish I could be a sponsor, maybe one day, maybe somehow!

Would you also support a pull request? Would you give a pointer to the area of the code best suited to this message introspection?

facontidavide, 2020-08-29:
sure, PR are welcome.

The place where this functionality should be added is **somehow**  here:

https://github.com/facontidavide/PlotJuggler/blob/7926f2119c543166e481f4ceb60b0ef0cb9313d2/plugins/ROS/ros1_parsers/ros1_parser.cpp#L32

This is the function called when a generic, unknown message is parsed and it uses the **ros_type_introspection** library.

A potential solution might be to extend ros_type_introspection itself, to do the conversion to RPY at the first pass and not later.

facontidavide, 2026-01-06:
closing stale issue