PlotJuggler/PlotJuggler #325: Crashes subscribing to some topics

erichahn, 2020-08-19:
[~]$ source torc/git/planning/mp_rviz_plugin/devel/setup.bash 
[~]$ rosrun plotjuggler PlotJuggler 
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadROS.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamROS.so" : is a DataStreamer plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libRosoutPublisher.so" : is a StatePublisher plugin
"libRosTopicPublisher.so" : is a StatePublisher plugin
terminate called after throwing an instance of 'RosIntrospection::RangeException'
  what():  Floating point truncated
Stack trace (most recent call last):
#31   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa6b0ea4, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#30   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa65e631, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool)
#29   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f20f98d98d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa65f65e, in QApplication::notify(QObject*, QEvent*)
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa65783b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa7da6fa, in QMenu::event(QEvent*)
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa696047, in QWidget::event(QEvent*)
#24   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa7d83d2, in QMenu::mouseReleaseEvent(QMouseEvent*)
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa7d75aa, in QMenu::leaveEvent(QEvent*)
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa7d00fb, in QMenu::setIcon(QIcon const&)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa65380b, in QAction::activate(QAction::ActionEvent)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f20fa651121, in QAction::triggered(bool)
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f20f990866e, in QMetaObject::activate(QObject*, int, int, void**)
#18   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x5632fdb4b137, in main
#17   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x5632fdb4acc1, in main
#16   Object "/opt/ros/melodic/lib/plotjuggler/libDataStreamROS.so", at 0x7f20a8e875e9, in DataStreamROS::start(QStringList*)
#15   Object "/opt/ros/melodic/lib/plotjuggler/libDataStreamROS.so", at 0x7f20a8e863b2, in DataStreamROS::extractInitialSamples()
#14   Object "/opt/ros/melodic/lib/libroscpp.so", at 0x7f20c14ac61a, in ros::CallbackQueue::callAvailable(ros::WallDuration)
#13   Object "/opt/ros/melodic/lib/libroscpp.so", at 0x7f20c14aa8a8, in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)
#12   Object "/opt/ros/melodic/lib/libroscpp.so", at 0x7f20c14fffa1, in ros::SubscriptionQueue::call()
#11   Object "/opt/ros/melodic/lib/plotjuggler/libDataStreamROS.so", at 0x7f20a8e97681, in ros::SubscriptionCallbackHelperT<boost::shared_ptr<RosIntrospection::ShapeShifter const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&)
#10   Object "/opt/ros/melodic/lib/plotjuggler/libDataStreamROS.so", at 0x7f20a8e8cce1, in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<RosIntrospection::ShapeShifter const> const&)>, void, boost::shared_ptr<RosIntrospection::ShapeShifter const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<RosIntrospection::ShapeShifter const>)
#9    Object "/opt/ros/melodic/lib/plotjuggler/libDataStreamROS.so", at 0x7f20a8e85871, in DataStreamROS::topicCallback(boost::shared_ptr<RosIntrospection::ShapeShifter const> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#8    Object "/opt/ros/melodic/lib/plotjuggler/libDataStreamROS.so", at 0x7f20a8e9dfdb, in CompositeParser::parseMessage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, nonstd::span_lite::span<unsigned char, -1l>, double)
#7    Object "/opt/ros/melodic/lib/plotjuggler/libDataStreamROS.so", at 0x7f20a8e9e2c5, in IntrospectionParser::parseMessage(nonstd::span_lite::span<unsigned char, -1l>, double)
#6    Object "/opt/ros/melodic/lib/plotjuggler/libDataStreamROS.so", at 0x7f20a8ea1fb6, in double RosIntrospection::Variant::convert<double>() const
#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f20f9358d53, in __cxa_throw
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f20f9358b20, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f20f9358ae5, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f20f9352956, in 
#1    Source "/build/glibc-2ORdQG/glibc-2.27/stdlib/abort.c", line 79, in __GI_abort [0x7f20f895f8b0]
#0    Source "/build/glibc-2ORdQG/glibc-2.27/signal/../sysdeps/unix/sysv/linux/raise.c", line 51, in __GI_raise [0x7f20f895df47]
Aborted (Signal sent by tkill() 10395 1000)
Aborted (core dumped)

using

facontidavide, 2020-08-19:
You know I am going to ask you for a rosbag to reproduce this, don't you?

erichahn, 2020-08-19:
I do thanks to your wonderful issue templates - it has been sent to you.
When playing the bag and subscribing to it from plotjuggler the issue is present.

When attempting to load the entire bag it complains "Floating point truncated"
Older versions of plot juggler supported this bag. If I knew what version it was I would attach that here.

I am on 18.04 melodic if that makes a difference.

facontidavide, 2020-08-20:
Ok, mystery solved. The error is exactly what it says it is.

You are converting a very large long integer to double and there is numerical cancelation in the process.

Therefore, the value you will see in PlotJuggler WILL be slightly misleading. **False data is worse than no data at all.**

A simple solution would be to convert your number, currently expressed in nanoseconds to microseconds on your side.

https://en.wikipedia.org/wiki/Double-precision_floating-point_format

This is one of the incriminated numbers:

        1.597.865.833.630.083.120 unsigned long

As long as your big number can be expressed in 52 bits (roughly 4 * 10^15), you are safe from numerical cancelation.

facontidavide, 2020-08-20:
Said that, I need a better message in the UI ;)

closed 2020-08-20 by facontidavide