PlotJuggler/PlotJuggler #328: No plugin loaded to process a data file (foxy)

PGotzmann, 2020-09-16:
First of all thank you for this awesome tool, which has saved me a lot of time so far.
I am glade you found the time to port this tool to ROS foxy, although I am not sure if the port is entirely finished since the related question #307 is still an open issue.
So please excuse me if this is still a work in progress and not jet supposed to work.

I have tried to build the `foxy` branch of Plotjugler from source using colcon as described in the readme. However when I run PlotJuggler using `rosrun` and try to load a data file (File/Load Data) I get a warning saying _"No plugin was loaded to process a data file"_.

The colcon build finished without any erros, the underlay/overlay should be sources correctly and I see the compiled shared object files of the plugins in `install/plotjuggler/lib/plotjuggler`.
Is there anything that I am missing here?

[image]

facontidavide, 2020-09-16:
can you please check the output in the console? It should print the list of loaded plugins.
Also, tell me which files are in that folder.

facontidavide, 2020-09-16:
wait...
are you using the foxy branch of the repository?

PGotzmann, 2020-09-16:
Thank you for your reply and taking the time to look into this.

I have added a list of the plugins located at `install/plotjuggler/lib/plotjuggler` to my original post.
Unfortunately I am unable get the log output you requested.  If I run PlotJuggler using `ros2 run plotjuggler PlotJuggler` or also if i run the PlotJuggler binary directly I do not get any log output at all. The application just starts up without anything printed to the terminal. Is there a verbose setting or something like that that I am missing?

I also double checked the branch and rebuild in a clean workspace just to be sure.

facontidavide, 2020-09-16:
>I have added a list of the plugins located at install/plotjuggler/lib/plotjuggler to my original post.

yes you did, sorry.

> I am using the foxy branch. Am I supposed to use the master?

foxy branch is the correct one (master won't compile in foxy).

It must be some kind of issue with installation directories, I have to figure that out.

facontidavide, 2020-09-17:
problem solved

closed 2020-09-17 by facontidavide

PGotzmann, 2020-09-18:
Thanks again for taking the time to look into this issue.
I can confirm that is working for me now.

Unfortunately I ran yet into another issue. However, they appear to be unrelated to me so I opened a new Issue for that.