PlotJuggler/PlotJuggler #329: Segfault parsing odom message from rosbag2 file (foxy)

PGotzmann, 2020-09-18:
branch: foxy (24010b7)

When trying to load `nav_msgs/msg/Odometry` message data from a `rosbag2` bagfile in PlotJuggler, a segmentation fault occurs while parsing the messages.
I have included a small bagfile that reproduces the issue for me as well as the full log output.
The bagfile can be played without any issues with `ros2 bag play`, so it probably is not corrupted.

**Bagfies and log output:**
[rosbag2_odom_crash.zip](https://github.com/facontidavide/PlotJuggler/files/5244552/rosbag2_odom_crash.zip)
[backtrace_odom_crash.log](https://github.com/facontidavide/PlotJuggler/files/5244576/backtrace_odom_crash.log)

facontidavide, 2020-09-18:
I think it is fixed. can you try the last commit?

PGotzmann, 2020-09-18:
Thank you, the issue seems to be gone now.
Closing this ticket.

closed 2020-09-18 by PGotzmann