PlotJuggler/PlotJuggler #333: Visual issue in dark mode: Placeholder for filter

g3n35i5, 2020-09-25:
Just a small visual problem: In dark mode the placeholder for the filter is displayed in black and therefore cannot be recognized easily. As soon as text is entered, the color changes to white.
[image]
[image]

g3n35i5, 2020-09-25:
I don't know why but after rebuilding and restarting the application a few times this magically disappeared (for now). Maybe it's just an issue with my environment.

facontidavide, 2020-09-29:
hi. Thanks for reporting. Right now I am focused on PlotJuggler 3.0 and the new dark filter there. I will keep this in mind.

closed 2020-09-29 by facontidavide