PlotJuggler/PlotJuggler #334: Enhancement : plot distribution of data

simchanu29, 2020-09-29:
I am currently working on some data from our robotic experiment. I currently use external tools to plot the distribution of the time between each log in our system. 

I've not managed to create a script to compute this distribution inside plotjuggle because this is a plot with a specific X axis.

It would be nice to have some way to plot the distribution of the data, or some option to create a XY plot from a script.

facontidavide, 2020-09-29:
I do not understand what you mean, sorry. What is the "distribution of the time between each log" ?

XY are well supported when you have a (t,X) and (t,Y) timeseries, sharing the same time axis and you want to build XY from them

simchanu29, 2020-09-30:
I think data distribution is the right word but i may be mistaken. For example, the result of a FFT is the distribution of the frequencies in a given signal.

For example I have a log with a timestamp variable then I compute the delta between 2 consecutive timestamp. The objective is to compute the frequency of appearance of each class of data. 
In practice, a point n of the result curve is the sum of every dt between n*step an (n+1)*step, n being the index of the point in the result curve and step being in that example 1ms.

I dont know any way to separate this XY plot in 2 timeseries since the final data distribution is totally independant from time.

Since plotjuggler seems to handle all data as timeseries, maybe it would need some major changes. I've done a search in the already closed issues. It seems mine is similar to #91 in the sense that a FFT compute a distribution as well.

facontidavide, 2020-10-02:
That would require indeed major changes. Of course that would accommodate many similar requests.

facontidavide, 2020-10-02:
Just a question. Would you or someone reading this pay for a professional version of PJ with this feature?

simchanu29, 2020-10-05:
Since my company already use a similar software called imc FAMOS, that unfortunately doesn't run on linux nor is compatible with ROS, I'm not sure my company would pay for such a feature. At least for now.

But a professional product of the same quality of imc FAMOS but running on Windows and working closely with ROS would certainly be attractive enough to pay for it.

facontidavide, 2020-10-05:
Thanks a lot for sharing this information. I am having a look to FAMOS :wink: 

Coming back to your problem, the short answer is: this is not going to be possible for now even if it can be technically done in the future if I manage to get some sponsor(s).

closed 2020-10-05 by facontidavide