PlotJuggler/PlotJuggler #337: Question: How to group values in the variable explorer

g3n35i5, 2020-10-02:
I would like to know if and if yes how it is possible to group array values (e.g. a server responds with an array of values for a variable) in the variable explorer like so:

```
MyVariable
   0
   1
   2
   3
   ...
```

or even better

```
MyVariable
   namedValue1
   namedValue2
   namedValue3
   namedValue4
   ...
```

Currently, I'm adding my variables to the plot like this:

```c++
void MyClass::registerToDataMap(PlotDataMapRef& map) const
{
  if (this->isArray())
  {
    for (size_t i = 0; i < this->arrayLength_; i++)
    {
      map.addNumeric(this->getName() + "[" + std::to_string(i) + "]");
    }
  }
  else
  {
    map.addNumeric(this->getName());
  }
}
```

which results in this:
[image]

facontidavide, 2020-10-02:
The hierarchy in the tree is automatically created by the character "/"

closed 2020-10-02 by facontidavide

g3n35i5, 2020-10-02:
Thank you very much! Did I miss a section in the documentation or is this behavior not documented?

facontidavide, 2020-10-02:
I think 🤔 it is ironic since I have zero documentation about hot to create plugins😝

g3n35i5, 2020-10-02:
Nice one :D Okay thanks!