PlotJuggler/PlotJuggler #338: Question: Does PlotJuggler support value units

g3n35i5, 2020-10-09:
Is it possible to define the unit of values and display them on the axes accordingly?

It would then be particularly interesting to know whether the basic unit (e.g. voltage) can be defined and the corresponding conversion and labeling in other powers of ten (mV, µV, ...) is possible.

Thank you for all your great work on PlotJuggler!

facontidavide, 2020-10-09:
no, for the time being I am not planning to add this feature.

First of all, where should this information, i.e. the units, comes from? Technically adding a axis title is very easy, but it is more a problem of data format.

How do you load your data?

g3n35i5, 2020-10-09:
I am currently working on a plugin, but unfortunately I cannot give you more information at this time.
There might be a list of supported units defined in PlotJuggler and if a line is tagged with a certain string (e.g. PlotJugglerUnits.Voltage), the unit will be shown in the plot. This feature, if implemented, is only optional, so the normal workflow should not be affected at all.

facontidavide, 2020-10-09:
> I am currently working on a plugin, but unfortunately I cannot give you more information at this time.

In that case, I can not give you more answers at this time :slightly_smiling_face: 

> There might be a list of supported units defined in PlotJuggler and if a line is tagged with a certain string (e.g. PlotJugglerUnits.Voltage), the unit will be shown in the plot. 

I would NOT do that, but rather make it an optional property of **PlotData** itself.

What do you want to happen if you drag and drop on the same plotting area two timeseries with different type (voltage and kilograms) or different scale (meters / millimeters)?

I think you are thinking about the implications at the UX level very thoughtfully.

Consider also that I am releasing the new PlotJuggler 3 in 1 month and the plugins will need some (not much) refactoring.

closed 2020-10-09 by facontidavide

g3n35i5, 2020-10-09:
> What do you want to happen if you drag and drop on the same plotting area two timeseries with different type (voltage and kilograms) or different scale (meters / millimeters)? I think you are thinking about the implications at the UX level very thoughtfully.

You're right, this is indeed a point which has to be considered, maybe at the point where multiple units are needed in the plot, the unit information can be moved into a tooltip information which is available on hovering the datapoints.

> Consider also that I am releasing the new PlotJuggler 3 in 1 month and the plugins will need some (not much) refactoring.

Thanks for the info, i'm really looking forward to the new version :)