PlotJuggler/PlotJuggler #344: Time delay in graph

DimitriEckert137, 2020-11-18:
Hi!

First of all thank for the great work with plotjuggler, its really handy.
I ran into the problem that some of my measurements are displayed in a kind of delayed fashion. I attached the bag file with the measurements. Specifically the topic "log" with all its subtopics is displayed in plotjuggler ahead of the other measurements. The measurements are displayed at the same "time-position" in the graph but when I look at the table on the left hand side and the actual values of the timestamps then I see they are different. At the start its about 0.5s towards the end its about 4s. Is there a way to prevent this?

[2020-11-18-08-36-55.zip](https://github.com/facontidavide/PlotJuggler/files/5561094/2020-11-18-08-36-55.zip)

facontidavide, 2020-11-18:
Hi,

thanks for reporting, I will investigate

facontidavide, 2020-11-24:
By default the Header timestamp is NOT used, but the time in the rosbag, that is equivalent to the time when **rosbag play** received the message.

Use the checkbox on top to change that:

[image]

closed 2020-11-24 by facontidavide

DimitriEckert137, 2020-11-24:
Oh i did not see that. Thanks a lot for clearing it up!