PlotJuggler/PlotJuggler #345: Installation of PlotJuggler on ros2 eloquent doesn't work

yangli-flexiv, 2020-11-20:
I have ros2 eloquent installed in my computer. But when I tried this following installation command, it didn't work:

sudo apt-get install ros-eloquent-plotjuggler.

Can you please check whether PlotJuggler works with Eloquent? I think maybe this above command needs to be tested.

facontidavide, 2020-11-23:
In the future, I am not going to support any version older than Foxy, at least **not in my spare time**.

If you or anybody want me to support other version, consider becoming a Sponsor.

https://github.com/sponsors/facontidavide

facontidavide, 2020-12-24:
Eloquent is EOL. porting PlotJuggler 3 there will never happen unless sponsored. Move to Foxy :stuck_out_tongue:

closed 2020-12-24 by facontidavide