PlotJuggler/PlotJuggler #346: Cannot load .bag files on Windows version

AlessandroMelino, 2020-11-23:
Hello.

I have installed the [Windows version of PlotJuggler](https://github.com/facontidavide/PlotJuggler/releases/download/2.8.4/PlotJuggler_WIN_Installer-2.8.4.exe) and when I run it and try to load a .bag file, there is no option to do that, just .csv and .ulg files. I saw videos of people loading .bag files, so I supose it is possible but I don't know how.

By the way, I installed it in my Ubuntu PC through apt-get, and when I run it with rosrun command an error appears:
`Segmentation fault (core dumped)
`
I hope anyone can help me.

Best regards.
Alessandro

facontidavide, 2020-11-23:
**The windows version DOES NOT support ROS**

The error on Ubuntu is worse. I am not sure how to reproduce this, since it seems to work for everybody else.

flaavour, 2020-11-23:
I guess I'm facing the same Ubuntu issue: 
```shell
Stack trace (most recent call last):
#22   Object "", at 0xffffffffffffffff, in 
#21   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55b1664ef759, in main
#20   Source "/build/glibc-S7xCS9/glibc-2.27/csu/../csu/libc-start.c", line 310, in __libc_start_main [0x7fbc110b1bf6]
#19   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55b1664eeef8, in main
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbc120a388e, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#17   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fbc0f31c6db, in g_main_context_iteration
#16   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fbc0f31c64f, in g_main_context_dispatch
#15   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fbc0f31c416, in g_main_context_dispatch
#14   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbc120a4262, in QEventDispatcherGlibPrivate::runTimersOnceWithNormalPriority()
#13   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbc1204d04c, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#12   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbc1204a8d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#11   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbc12dd0103, in QApplication::notify(QObject*, QEvent*)
#10   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fbc12dc883b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#9    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbc1207a0c1, in QObject::event(QEvent*)
#8    Object "/usr/lib/x86_64-linux-gnu/libQt5Network.so.5", at 0x7fbc13847d48, in QHttpMultiPart::qt_metacall(QMetaObject::Call, int, void**)
#7    Object "/usr/lib/x86_64-linux-gnu/libQt5Network.so.5", at 0x7fbc137b775e, in QHttpPart::setBody(QByteArray const&)
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbc12079554, in QMetaObject::activate(QObject*, int, int, void**)
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Network.so.5", at 0x7fbc13753038, in QNetworkAccessManager::createRequest(QNetworkAccessManager::Operation, QNetworkRequest const&, QIODevice*)
#4    Object "/usr/lib/x86_64-linux-gnu/libQt5Network.so.5", at 0x7fbc1374ef83, in QNetworkAccessManager::supportedSchemes() const
#3    Object "/usr/lib/x86_64-linux-gnu/libQt5Network.so.5", at 0x7fbc1374e1a1, in QNetworkAccessManager::finished(QNetworkReply*)
#2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbc1207966e, in QMetaObject::activate(QObject*, int, int, void**)
#1    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x55b1664ff64e, in main
#0    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fbc11eebcbc, in QString::toInt(bool*, int) const
Segmentation fault (Signal sent by the kernel [(nil)])
/opt/ros/melodic/bin/plotjuggler: line 20: 14529 Segmentation fault      (core dumped) rosrun plotjuggler PlotJuggler
```
Maybe this should be another Github ticket?

AlessandroMelino, 2020-11-24:
Hello.

Thank you por your answer. It would be interesting if Windows version could load .bag files.

By the way, I updated the Ubuntu version and now it works.

Best regards.

closed 2020-11-24 by AlessandroMelino