PlotJuggler/PlotJuggler #347: Segmentation Fault

SDBonhof, 2020-11-23:
Hi!

I have installed plotjuggler using apt with

``` bash
$ sudo apt install ros-melodic-plotjuggler
```
Running the application with
``` bash
$ plotjuggler
``` 
shows the 'loading meme' very briefly, but then quits and shows the following error:

```
Stack trace (most recent call last):
#22   Object "", at 0xffffffffffffffff, in 
#21   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x5582c6210759, in main
#20   Source "/build/glibc-S7xCS9/glibc-2.27/csu/../csu/libc-start.c", line 310, in __libc_start_main [0x7f273cd71bf6]
#19   Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x5582c620fef8, in main
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f273dd6388e, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#17   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f273afdc6db, in g_main_context_iteration
#16   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f273afdc64f, in g_main_context_dispatch
#15   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f273afdc416, in g_main_context_dispatch
#14   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f273dd64262, in QEventDispatcherGlibPrivate::runTimersOnceWithNormalPriority()
#13   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f273dd0d04c, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#12   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f273dd0a8d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#11   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f273ea90103, in QApplication::notify(QObject*, QEvent*)
#10   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f273ea8883b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#9    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f273dd3a0c1, in QObject::event(QEvent*)
#8    Object "/usr/lib/x86_64-linux-gnu/libQt5Network.so.5", at 0x7f273f507d48, in QHttpMultiPart::qt_metacall(QMetaObject::Call, int, void**)
#7    Object "/usr/lib/x86_64-linux-gnu/libQt5Network.so.5", at 0x7f273f47775e, in QHttpPart::setBody(QByteArray const&)
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f273dd39554, in QMetaObject::activate(QObject*, int, int, void**)
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Network.so.5", at 0x7f273f413038, in QNetworkAccessManager::createRequest(QNetworkAccessManager::Operation, QNetworkRequest const&, QIODevice*)
#4    Object "/usr/lib/x86_64-linux-gnu/libQt5Network.so.5", at 0x7f273f40ef83, in QNetworkAccessManager::supportedSchemes() const
#3    Object "/usr/lib/x86_64-linux-gnu/libQt5Network.so.5", at 0x7f273f40e1a1, in QNetworkAccessManager::finished(QNetworkReply*)
#2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f273dd3966e, in QMetaObject::activate(QObject*, int, int, void**)
#1    Object "/opt/ros/melodic/lib/plotjuggler/PlotJuggler", at 0x5582c622064e, in main
#0    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f273dbabcbc, in QString::toInt(bool*, int) const
Segmentation fault (Signal sent by the kernel [(nil)])
/opt/ros/melodic/bin/plotjuggler: line 20: 25079 Segmentation fault      (core dumped) rosrun plotjuggler PlotJuggler
```

The application worked earlier today but stopped working after an attempt to install some dependencies for keras/tensorflow. The error message I get doesn't really help me which is why I am posting here.

I am using Ubuntu 18.04.

If you need any other information I'll be happy to provide it.

Thanks!

padhupradheep, 2020-11-23:
Hello, 

Same distro and same issue! Recently updated plotjuggler to the newest version!

SDBonhof, 2020-11-23:
I have found the same error at #346. Seems like something might have gone wrong with a recent update.

facontidavide, 2020-11-23:
Can you please compile the matest vesion in branch **main** and tell me if the problem persists?

flaavour, 2020-11-23:
Seems the latest **main** fixes the issue.

facontidavide, 2020-11-23:
great. Enjoy plotJuggler 3. 

The Debian package will take a couple of weeks to be released by OpenRobotics, in the midtime, the README has instruction to compile it yourself.

closed 2020-11-23 by facontidavide