PlotJuggler/PlotJuggler #348: PlotJuggler does not build with clang

timonegk, 2020-11-23:
It's me again with a clang build error!
```
.../PlotJuggler/./include/PlotJuggler/plotdata.h:509:3: error: unsupported: typedef changes linkage of anonymous type, but linkage was already computed
} PlotDataMapRef;
  ^
.../PlotJuggler/./include/PlotJuggler/plotdata.h:493:15: note: use a tag name here to establish linkage prior to definition
typedef struct
              ^
               PlotDataMapRef
1 error generated.
```
PlotJuggler: current master (3beeafa4f68ddc39e73a804b46d843a9a217ed90)
clang: 10.0.1-8+b1
OS: Debian Testing

facontidavide, 2020-11-23:
Thanks for reporting, I will check it ASAP

facontidavide, 2020-11-23:
Check if the last commit fix the issue

facontidavide, 2020-11-23:
It should be fixed, I think

closed 2020-11-23 by facontidavide

timonegk, 2020-11-23:
Yes, this issue is fixed. However, the build now fails with the following message:
```
/usr/bin/ld: .../catkin_ws/devel/.private/plotjuggler/lib/libpaho-mqtt3a.a(MQTTAsync.c.o): relocation R_X86_64_32 against `.bss' can not be used when making a shared object; recompile with -fPIC
```
It does build if I build `paho.mqtt.c` as a shared library (switch bools in `CMakeLists.txt:219,220` and remove `-static` suffix in `plugins/DataStreamMQTT/CMakeLists.txt:17,18`). I don't know though if that's a valid fix or if the library is built as static for a reason.