PlotJuggler/PlotJuggler #349: PlotJuggler does not load ROS plugins when built from source with catkin

ozzdemir, 2020-11-26:
Firstly, thanks for building/maintaining such a useful tool. I am using PlotJuggler with ROS from apt(v 2.8.4) and found out that version 3 is released and wanted to try. I installed the packages just described in [here](https://github.com/PlotJuggler/plotjuggler-ros-plugins#how-to-compile-plotjuggler-from-source) but PlotJuggler couldn't load the ROS plugins since it was looking for the source directory instead of the install directory.
`Loading compatible plugins from directory:  "/home/oguz/plotjuggler_ws/src/plotjuggler-ros-plugins"
Number of plugins loaded:  0 
`
In order to test, changed https://github.com/PlotJuggler/PlotJuggler/blob/a5e4ca661f23de89a8a7df2f0e6ceb283659ce5d/src/main.cpp#L119-L122
with
```
#ifdef COMPILED_WITH_CATKIN
    extra_path = QString::fromStdString(ros::package::getPath("plotjuggler_ros"));
    extra_path.replace("/src/plotjuggler-ros-plugins", "/devel/lib/plotjuggler_ros");
#endif
```
and it loaded plugins successfully. Seems like `ros::package::getPath` returns source directory instead of install directory.
Hope this issue can help.

facontidavide, 2020-11-26:
you are right. I need to fix this ASAP. Thanks for reporting

facontidavide, 2020-11-26:
I implemented a workaround. But I will work on a way to improve this, eventually.

closed 2020-11-26 by facontidavide